scs/smartmeter-datacollector #10: Clock & Device ID abstraction

aselviar, 2021-09-25:
When the clock or Device ID with obis code below are not pushed from the smart meter because they are configured not to push them (depends on the standard CII configuration of each DSO), the dlms parser does not parse the data and the data are not saved in the influx db for the demo. This was the case when we tried to read out the AEW smart meter with different configuration as that of EKZ. 

The reason is that the id and clock obis codes are hard coded. Possible work arounds:

- when the clock is not pushed, take the linux time instead
- if the device id is not pushed, add alternative obis id codes to choose from or hardcode as "smartmeter1", "smartmeter2" and so on

<div class="file-box blob-wrapper my-1" style="box-sizing: border-box; margin-top: 4px !important; margin-bottom: 4px !important; overflow: auto hidden; border: 1px solid var(--color-border-primary); border-radius: 6px; color: rgb(36, 41, 47); font-family: -apple-system, system-ui, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

id_obis="0.0.42.0.0.255",
--
93 | clock_obis="0.0.1.0.0.255",

</div><div class="d-flex flex-wrap text-small color-text-secondary" style="box-sizing: border-box; flex-wrap: wrap !important; font-size: 12px !important; display: flex !important; color: var(--color-fg-muted)  !important; font-family: -apple-system, system-ui, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="mr-3" style="box-sizing: border-box; margin-right: 16px !important;"><span class="" style="box-sizing: border-box;"><span class="repo-language-color" style="box-sizing: border-box; position: relative; top: 1px; display: inline-block; width: 12px; height: 12px; border: 1px solid var(--color-primer-border-contrast); border-radius: 50%; background-color: rgb(53, 114, 165);"></span></span></div></div><br class="Apple-interchange-newline">

tom-ch1, 2021-09-26:
See tom-ch1@7a3b34af2e27d1ad089d7c1fb1333c1445d98abb for the patch we used to make this work for an AEW smart meter (same brand and model as EKZ, just different config).

tom-ch1, 2021-09-26:
It would be useful to have configuration-style files to specify the properties of a certain model. The AEW smart meter, for instance, just uses a different obis id for the smart meter id.
Something like lge450_EKZ.conf or lge450_AEW.conf instead of lge450.py

aselviar, 2021-09-29:
@Montellese @tom-ch1 @raymar9 
I would not go with different configuration files for the different utilities, as there are too many to create a conf file for each one of them. 

A more abstract way to solve this is:

- clock: if no clock_obis is pushed from the smart meter --> use linux time instead
- for id_obis check if one of the following is pushed:

1. 0-0:96.1.0;2
2. 0-0:96.1.1;2
3. 0-0:96.1.2;2
4. 0-0:96.1.3;2
5. 0-0:96.1.4;2
6. 0-0:96.1.5;2

if none of them pushed --> generate a generic id

tom-ch1, 2021-09-29:
Fair enough, that's a good point, Angelos. One configuration file per utility company would definitely be an overkill.

However, a configuration file instead of a python file would still be preferrable. Imagine some user wants to use the library but something has to be adapted for his smart meter. He won't touch a python file, but he might adapt a config file if it is well documented.
As for the id: I don't quite understand the notation, but it seems to me that your proposal won't match neither EKZ nor AEW

aselviar, 2021-10-05:
Sorry, there was a confusion here.

What the EKZ Smart Meter pushes is the COSEM logical device name with the OBIS Code: "0.0.42.0.0.255"

The AEW Smart Meter does not push this but pushes insted Device ID 1: "0.0.96.1.0.255"

The notation is different in the code as in the OBIS structure since the OBIS Code is represented with this notation in the GuruX library (dot instead of -,:,; and 255 in the end).

As ID of the smart meter for the code I would search the following OBIS codes - if they are pushed - :

0-0:42.0.0;2 - COSEM logical device name
0-0:96.1.0;2 - Device ID 1 
0-0:96.1.1;2 - Device ID 2
0-0:96.1.2;2 - Device ID 3
0-0:96.1.3;2 - Device ID 4
0-0:96.1.4;2 - Device ID 5
0-0:96.1.5;2 - Device ID 6

If none of them is pushed, the create a generic ID. 

The above OBIS Codes are universal and should apply for all smart meters.