scs/smartmeter-datacollector #14: Hardware Setup e450 wired RJ12

maestrone, 2021-11-04:
A nice project, thank you for the code...
Please, can you tell the functional hardware setup for a wiring connection between Raspberry <> L&G E450 RJ12?

Regards

maestrone, 2021-11-16:
I found a solution:

TSS721 USB to MBus Slave Module

closed 2021-11-16 by maestrone

raymar9, 2021-11-29:
Hi maestrone
Sorry for the delayed response. Thank you for the interest in our project!
Yes, as you found out yourself, you can use any USB to M-Bus-Slave dongle. I updated the [Wiki ](https://github.com/scs/smartmeter-datacollector/wiki) with the model we used, however it was quite expensive.

novoid, 2022-04-16:
Hi,
This dongle is quite expensive ([CHF 220](https://zeta-eng.ch/fileadmin/zeta/images/PDF%20Dateien/USB_interfaces_-_price_list.pdf)) and hard to get.

I've got multiple RJ12 cables from the good old FAX epoch. Is it possible to connect the PIN 3 and 4 of such a cut FAX wire to the GPIO ports of a RPi and use this project to collect its data somehow?

raymar9, 2022-04-20:
No, that would not work. The dongle also contains the circuitry for the M-Bus connectivity which the Raspberry Pi doesn't provide.