scs/smartmeter-datacollector #18: Uncomplete data collection

maestrone, 2021-12-06:
hello togheter

I try to catch data from a E450 whit your software. I received just 9 records (some are not correct). Any idea how I can fix this?

Regards

M

`WARNING:root:No config file found. Using default config.
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/ttyUSB1'.
INFO:sink:Connected to MQTT broker.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F3 0C 07 E5 0C 06 01 08 33 37 FF 80 00 00 02 0B 01 0B 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 00 60 01 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 01 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 02 07 00 FF 0F 02 12 00 00 9E 9A 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F3 0C 07 E5 0C 06 01 08 33 37 FF 80 00 00 02 0B 01 0B 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 00 60 01 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 01 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 02 07 00 FF 0F 02 12 00 00 9E 9A 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 02 04 12 00 03 09 06 01 01 01 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 02 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 05 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 1F 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 33 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 47 07 00 FF 0F 02 12 00 00 09 06 00 08 19 09 00 FF 09 08 35 35 38 39 C7 B6 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F3 0C 07 E5 0C 06 01 08 33 37 FF 80 00 00 02 0B 01 0B 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 00 60 01 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 01 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 02 07 00 FF 0F 02 12 00 00 9E 9A 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 02 04 12 00 03 09 06 01 01 01 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 02 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 05 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 1F 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 33 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 47 07 00 FF 0F 02 12 00 00 09 06 00 08 19 09 00 FF 09 08 35 35 38 39 C7 B6 7E 7E A0 37 CE FF 03 13 5A C8 E0 C0 00 03 00 00 26 33 32 32 37 06 00 00 05 CE 06 00 00 00 00 06 00 86 2F 2B 06 00 01 76 F4 06 00 02 EA 27 12 00 3F 12 02 94 12 00 09 48 16 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
DEBUG:smartmeter:40 0.8.25.9.0.255 1: 0.8.25.9.0.255
DEBUG:smartmeter:1 0.0.96.1.0.255 2: Meter_ID
DEBUG:smartmeter:3 1.0.1.7.0.255 2: 1486
DEBUG:smartmeter:3 1.0.2.7.0.255 2: 0
DEBUG:smartmeter:3 1.1.1.8.0.255 2: 8793899
DEBUG:smartmeter:3 1.1.2.8.0.255 2: 95988
DEBUG:smartmeter:3 1.1.5.8.0.255 2: 191015
DEBUG:smartmeter:3 1.0.31.7.0.255 2: 63
DEBUG:smartmeter:3 1.0.51.7.0.255 2: 660
DEBUG:smartmeter:3 1.0.71.7.0.255 2: 9
DEBUG:smartmeter:ID Meter_ID found with OBIS code 0.0.96.1.0.255.
DEBUG:smartmeter:No clock object found at 0.0.1.0.0.255. Using system time.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Active Power +: 1486.0 W
DEBUG:sink:{"value": 1486.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Active Power -: 0.0 W
DEBUG:sink:{"value": 0.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Active Energy +: 8793899.0 Wh
DEBUG:sink:{"value": 8793899.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Active Energy -: 95988.0 Wh
DEBUG:sink:{"value": 95988.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Reactive Energy +Ri Q1: 191015.0 VAh
DEBUG:sink:{"value": 191015.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Current L1: 63.0 A
DEBUG:sink:{"value": 63.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Current L2: 660.0 A
DEBUG:sink:{"value": 660.0, "timestamp": 1638773542} sent to MQTT broker.
INFO:DataLogger:Meter_ID - 2021-12-06T07:52:22.682119 - Current L3: 9.0 A
DEBUG:sink:{"value": 9.0, "timestamp": 1638773542} sent to MQTT broker.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F4 0C 07 E5 0C 06 01 08 34 00 FF 80 00 00 02 0E 01 0E 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 01 12 00 00 02 04 12 00 48 09 06 00 01 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 02 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 03 18 01 00 FF 0F 06 12 00 00 7D 36 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F4 0C 07 E5 0C 06 01 08 34 00 FF 80 00 00 02 0E 01 0E 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 01 12 00 00 02 04 12 00 48 09 06 00 01 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 02 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 03 18 01 00 FF 0F 06 12 00 00 7D 36 7E 7E A0 7D CE FF 03 13 D0 45 E0 40 00 02 00 00 6C 02 04 12 00 04 09 06 00 01 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 02 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 03 18 02 01 FF 0F 02 12 00 00 02 04 12 00 48 09 06 00 01 18 01 00 FF 0F 09 12 00 00 02 04 12 00 48 09 06 00 02 18 01 00 FF 0F 09 12 00 00 02 04 12 00 48 09 06 00 03 18 01 00 FF 0F 09 12 00 00 D8 C3 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F4 0C 07 E5 0C 06 01 08 34 00 FF 80 00 00 02 0E 01 0E 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 01 12 00 00 02 04 12 00 48 09 06 00 01 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 02 18 01 00 FF 0F 06 12 00 00 02 04 12 00 48 09 06 00 03 18 01 00 FF 0F 06 12 00 00 7D 36 7E 7E A0 7D CE FF 03 13 D0 45 E0 40 00 02 00 00 6C 02 04 12 00 04 09 06 00 01 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 02 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 03 18 02 01 FF 0F 02 12 00 00 02 04 12 00 48 09 06 00 01 18 01 00 FF 0F 09 12 00 00 02 04 12 00 48 09 06 00 02 18 01 00 FF 0F 09 12 00 00 02 04 12 00 48 09 06 00 03 18 01 00 FF 0F 09 12 00 00 D8 C3 7E 7E A0 85 CE FF 03 13 56 80 E0 C0 00 03 00 00 74 02 04 12 00 04 09 06 00 01 18 02 01 FF 0F 03 12 00 00 02 04 12 00 04 09 06 00 02 18 02 01 FF 0F 03 12 00 00 02 04 12 00 04 09 06 00 03 18 02 01 FF 0F 03 12 00 00 09 06 00 0B 19 09 00 FF 06 00 00 00 00 06 00 00 00 00 06 00 00 00 00 05 00 00 00 00 05 00 00 00 00 05 00 00 00 00 11 00 11 00 11 00 02 02 0F FD 16 0E 02 02 0F FD 16 0E 02 02 0F 00 16 FF 9E FC 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
DEBUG:smartmeter:40 0.11.25.9.0.255 1: 0.11.25.9.0.255
DEBUG:smartmeter:72 0.1.24.1.0.255 6: 0
DEBUG:smartmeter:72 0.2.24.1.0.255 6: 0
DEBUG:smartmeter:72 0.3.24.1.0.255 6: 0
DEBUG:smartmeter:4 0.1.24.2.1.255 2: 0
DEBUG:smartmeter:4 0.2.24.2.1.255 2: 0
DEBUG:smartmeter:4 0.3.24.2.1.255 2: 0
DEBUG:smartmeter:72 0.1.24.1.0.255 9: 0
DEBUG:smartmeter:72 0.2.24.1.0.255 9: 0
DEBUG:smartmeter:72 0.3.24.1.0.255 9: 0
DEBUG:smartmeter:4 0.1.24.2.1.255 3: [0.001, <Unit.CORRECTED_VOLUME: 14>]
DEBUG:smartmeter:4 0.2.24.2.1.255 3: [0.001, <Unit.CORRECTED_VOLUME: 14>]
DEBUG:smartmeter:4 0.3.24.2.1.255 3: [1.0, <Unit.NO_UNIT: 255>]
DEBUG:smartmeter:No clock object found at 0.0.1.0.0.255. Using system time.
DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 38 48 F5 0C 07 E5 0C 06 01 08 34 05 FF 80 00 00 02 0B 01 0B 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 00 60 01 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 01 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 02 07 00 FF 0F 02 12 00 00 C2 C3 7E
[... cut ...]
DEBUG:smartmeter:4 0.1.24.2.1.255 2: 0
DEBUG:smartmeter:4 0.2.24.2.1.255 2: 0
DEBUG:smartmeter:4 0.3.24.2.1.255 2: 0
DEBUG:smartmeter:72 0.1.24.1.0.255 9: 0
DEBUG:smartmeter:72 0.2.24.1.0.255 9: 0
DEBUG:smartmeter:72 0.3.24.1.0.255 9: 0
DEBUG:smartmeter:4 0.1.24.2.1.255 3: [0.001, <Unit.CORRECTED_VOLUME: 14>]
DEBUG:smartmeter:4 0.2.24.2.1.255 3: [0.001, <Unit.CORRECTED_VOLUME: 14>]
DEBUG:smartmeter:4 0.3.24.2.1.255 3: [1.0, <Unit.NO_UNIT: 255>]
DEBUG:smartmeter:No clock object found at 0.0.1.0.0.255. Using system time.
`

raymar9, 2021-12-06:
Hi maestrone

Some data is successfully parsed: those that are forwarded to MQTT. Is something wrong with those values? Are the current values plausible (660A seams too high)? Can you provide more information about your setup? I assume this is not a meter installed by EKZ?

There are some other OBIS codes later which are not parsed by the software (like OBIS 0.1.24.1.0.255). At the moment, we only parse a handful (most common) OBIS data fields. You can find the list of them [here](https://github.com/scs/smartmeter-datacollector/blob/master/smartmeter_datacollector/smartmeter/lge450.py#L20).

Further important note: Our software cannot control what OBIS records the smart meters send. The software only parses what it gets. If you want to receive more records, you need to configure this on your meter with a separate software.

maestrone, 2021-12-08:
thx for your replay 

My energy distributor (in Zürcher Oberland) gave me this specs:

Logischer Name | Register (every 5 sec)
0-8:25.9.0; 2 | Object list push setup consumer information 1
0-8:25.9.0; 1 | Logical name push setup consumer information 1
0-0:96.1.0; 2 | Device ID 1 (utility serial number 1, ID 2.1)
1-0:1.7.0; 2 | Active power import +P
1-0:2.7.0; 2 | Active power export –P
1-1:1.8.0; 2 | Active energy import +A (QI+QIV)
1-1:2.8.0; 2 | Active energy export –A (QII+QIII)
1-1:5.8.0; 2 | Reactive energy +Ri (QI)
1-0:31.7.0; 2 | Current L1
1-0:51.7.0; 2 | Current L2
1-0:71.7.0; 2 | Current L3

Logischer Name | Register (every minute)
0-9:25.9.0; 2 | Object list push setup consumer information 2
0-9:25.9.0; 1 | Logical name push setup consumer information 2
1-1:1.8.1; 2 | Active energy import +A (QI+QIV) rate 1
1-1:1.8.2; 2 | Active energy import +A (QI+QIV) rate 2
1-1:2.8.1; 2 | Active energy export -A (QII+QIII) rate 1
1-1:2.8.2; 2 | Active energy export –A (QII+QIII) rate 2
1-1:5.8.1; 2 | Reactive energy +Ri (QI) rate 1
1-1:5.8.2; 2 | Reactive energy +Ri (QI) rate 2

I change/add on the lge450.py:
RegisterCosem("1.1.1.8.1.255", MeterDataPointTypes.ACTIVE_ENERGY_P.value),
RegisterCosem("1.1.1.8.2.255", MeterDataPointTypes.ACTIVE_ENERGY_P.value),
RegisterCosem("1.1.2.8.1.255", MeterDataPointTypes.ACTIVE_ENERGY_N.value),
RegisterCosem("1.1.2.8.2.255", MeterDataPointTypes.ACTIVE_ENERGY_P.value),

and a get (every minute):
2021-12-08T10:46:38.066092 - Active Energy +: 2858900.0 Wh
2021-12-08T10:46:38.066092 - Active Energy +: 6008371.0 Wh
2021-12-08T10:46:38.066092 - Active Energy -: 85713.0 Wh
2021-12-08T10:46:38.066092 - Active Energy +: 10275.0 Wh

seems to me that its works - this is ok/enough for me

Regards

closed 2021-12-13 by raymar9