scs/smartmeter-datacollector #21: 'int' object is not subscriptable

LichtiMC, 2022-01-09:
Hello there,

I'm having the problem, that the datacollector service is restarting after every packet transmitted by the smartmeter with the error-message mentioned in the title.

Here is the full log of one restart-cycle:
`Jän 09 23:27:53 ha-server systemd[1]: python3-smartmeter-datacollector.service: Main process exited, code=exited, status=1/FAILURE
Jän 09 23:27:53 ha-server systemd[1]: python3-smartmeter-datacollector.service: Failed with result 'exit-code'.
Jän 09 23:27:53 ha-server systemd[1]: python3-smartmeter-datacollector.service: Service RestartSec=100ms expired, scheduling restart.
Jän 09 23:27:53 ha-server systemd[1]: python3-smartmeter-datacollector.service: Scheduled restart job, restart counter is at 9.
Jän 09 23:27:53 ha-server systemd[1]: Stopped Smart Meter Data Collector Service.
Jän 09 23:27:53 ha-server systemd[1]: Started Smart Meter Data Collector Service.
Jän 09 23:27:54 ha-server smartmeter-datacollector[20895]: DEBUG:asyncio:Using selector: EpollSelector
Jän 09 23:27:54 ha-server smartmeter-datacollector[20895]: INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0'.
Jän 09 23:27:54 ha-server smartmeter-datacollector[20895]: INFO:sink:Connected to MQTT broker.
Jän 09 23:28:01 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 78 1F D0 82 01 03 30 00 00 A4 1E AE 60 03 2D 12 96 F2 D2 86 5D 27 94 F3 59 34 7E
Jän 09 23:28:01 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jän 09 23:28:01 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 78 1F D0 82 01 03 30 00 00 A4 1E AE 60 03 2D 12 96 F2 D2 86 5D 27 94 F3 59 34 7E 91 AE FF E2 BC 1F 37 C0 7B 5C 30 A6 9A 48 89 06 30 97 97 8B EF 0D 55 0B BA 69 85 D6 77 61 D9 C3 72 DF 3C 7B F3 3F A3 CB A6 F8 3D B5 5A 71 BE DF 3A 4D B2 9A 89 BC C3 A1 89 11 E6 FF F2 09 4D BF 1F 7B 33 67 A9 F9 78 F4 90 9A 62 03 4E 3D 6E FD 3B 2E 0D 56 66 44 AC 7E
Jän 09 23:28:01 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 78 1F D0 82 01 03 30 00 00 A4 1E AE 60 03 2D 12 96 F2 D2 86 5D 27 94 F3 59 34 7E 91 AE FF E2 BC 1F 37 C0 7B 5C 30 A6 9A 48 89 06 30 97 97 8B EF 0D 55 0B BA 69 85 D6 77 61 D9 C3 72 DF 3C 7B F3 3F A3 CB A6 F8 3D B5 5A 71 BE DF 3A 4D B2 9A 89 BC C3 A1 89 11 E6 FF F2 09 4D BF 1F 7B 33 67 A9 F9 78 F4 90 9A 62 03 4E 3D 6E FD 3B 2E 0D 56 66 44 AC 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A B6 DB 45 4B A6 78 15 1F 0C 46 1D B9 7E
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 78 1F D0 82 01 03 30 00 00 A4 1E AE 60 03 2D 12 96 F2 D2 86 5D 27 94 F3 59 34 7E 91 AE FF E2 BC 1F 37 C0 7B 5C 30 A6 9A 48 89 06 30 97 97 8B EF 0D 55 0B BA 69 85 D6 77 61 D9 C3 72 DF 3C 7B F3 3F A3 CB A6 F8 3D B5 5A 71 BE DF 3A 4D B2 9A 89 BC C3 A1 89 11 E6 FF F2 09 4D BF 1F 7B 33 67 A9 F9 78 F4 90 9A 62 03 4E 3D 6E FD 3B 2E 0D 56 66 44 AC 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A B6 DB 45 4B A6 78 15 1F 0C 46 1D B9 7E 7C 7B 1C D6 A0 B9 0F 0D 03 5C F8 F2 D1 7B B7 9B D4 2A 45 DC F6 51 09 A2 FD 8B E6 6C 19 F1 89 90 C8 60 29 CE 62 27 20 BB B4 A4 F1 90 C6 47 C9 3A EB 44 65 B0 AE C3 20 E3 10 6E 28 E3 9C 57 E7 61 D6 70 01 53 DC 93 4D A0 C5 6C 74 80 32 2B 44 49 60 9C 22 2F A2 AC 60 C2 A5 1E 95 23 C2 DA 61 6C 60 14 8C 65 FE 97 F2 42 DE 95 3D 27 0D DE 69 7E
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 78 1F D0 82 01 03 30 00 00 A4 1E AE 60 03 2D 12 96 F2 D2 86 5D 27 94 F3 59 34 7E 91 AE FF E2 BC 1F 37 C0 7B 5C 30 A6 9A 48 89 06 30 97 97 8B EF 0D 55 0B BA 69 85 D6 77 61 D9 C3 72 DF 3C 7B F3 3F A3 CB A6 F8 3D B5 5A 71 BE DF 3A 4D B2 9A 89 BC C3 A1 89 11 E6 FF F2 09 4D BF 1F 7B 33 67 A9 F9 78 F4 90 9A 62 03 4E 3D 6E FD 3B 2E 0D 56 66 44 AC 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A B6 DB 45 4B A6 78 15 1F 0C 46 1D B9 7E 7C 7B 1C D6 A0 B9 0F 0D 03 5C F8 F2 D1 7B B7 9B D4 2A 45 DC F6 51 09 A2 FD 8B E6 6C 19 F1 89 90 C8 60 29 CE 62 27 20 BB B4 A4 F1 90 C6 47 C9 3A EB 44 65 B0 AE C3 20 E3 10 6E 28 E3 9C 57 E7 61 D6 70 01 53 DC 93 4D A0 C5 6C 74 80 32 2B 44 49 60 9C 22 2F A2 AC 60 C2 A5 1E 95 23 C2 DA 61 6C 60 14 8C 65 FE 97 F2 42 DE 95 3D 27 0D DE 69 7E 7E A0 30 CE FF 03 13 86 F8 E0 C0 00 03 00 00 1F 96 33 41 C0 6B 44 02 3C C0 2E 21 4F 56 B8 D8 7B 6C 73 17 20 37 62 4E 27 DE 9B 55 E4 49 35 37 FC D7 7E
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: DEBUG:smartmeter:DLMS packet complete and ready for parsing.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: INFO:root:App shutting down now.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: INFO:sink:Disconnected from MQTT broker.
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: Traceback (most recent call last):
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/bin/smartmeter-datacollector", line 14, in <module>
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     app.main()
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/app.py", line 69, in main
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     asyncio.run(build_and_start(app_config), debug=debug_mode)
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     return loop.run_until_complete(main)
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     return future.result()
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/app.py", line 29, in build_and_start
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     data_collector.process_queue())
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 46, in start
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     await self._serial.start_and_listen()
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/smartmeter/serial_reader.py", line 45, in start_and_listen
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     self._callback(data)
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 57, in _data_received
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     dlms_objects = self._parser.parse_to_dlms_objects()
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py", line 84, in parse_to_dlms_objects
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     parsed_objects = self._client.parsePushObjects(self._dlms_data.value[0])
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:   File "/usr/local/lib/python3.7/dist-packages/gurux_dlms/GXDLMSClient.py", line 1257, in parsePushObjects
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]:     classID = tmp[0]
Jän 09 23:28:02 ha-server smartmeter-datacollector[20895]: TypeError: 'int' object is not subscriptable
Jän 09 23:28:04 ha-server systemd[1]: python3-smartmeter-datacollector.service: Main process exited, code=exited, status=1/FAILURE
Jän 09 23:28:04 ha-server systemd[1]: python3-smartmeter-datacollector.service: Failed with result 'exit-code'.
`
Anybody knows of a solution?
Thank you.

raymar9, 2022-01-11:
Hi LichtiMC
This seems to be a bug in the converter library (from Gurux) that we use for HDLC/DLMS translation. Unfortunately your raw data (HDLC Buffer before "DLMS packet complete and ready for parsing") is encrypted with the key you probably entered in the configuration and we cannot reproduce your experienced error nor test whether the translation works with an updated version of the Gurux library.

LichtiMC, 2022-01-11:
I could provide you with the key.
How can I contact you?
Thank you.

raymar9, 2022-01-11:
Please send your key to this temporary mail address: 
pobawoc989@vinopub.com
Thanks for the idea of using `/dev/serial/by-id/..` for the devices (seen in your posted log). I had problems distinguishing the devices `/dev/tty*` when more than one adapter is attached.

LichtiMC, 2022-01-12:
Forgot to ping back:
Mail sent!
Thank you!

raymar9, 2022-01-12:
Sorry, something did not work with the temp mail service. I didn't receive your mail. Please send it again to the following address:
xrb4rzmk6qs@temp.mailbox.org
This is a better provider and should work. Thank you!

LichtiMC, 2022-01-12:
Mail sent.

raymar9, 2022-01-13:
Thanks, I received it. You will here from me as soon as I discover something.

raymar9, 2022-01-20:
Hi LichtiMC
I tried a couple of things with your raw data and the key that you sent me. Unfortunately nothing has worked so far.
I've written a unit test with your data and key to reproduce the behavior and to use step-by-step debugging. It seems that the data can not be decoded into the DLMS data objects (it is still bytecode after the tried conversion).
Also I used the Gurux DLMS Translator (online tool and Windows-Tool) to decode the data, but without success.
I suspect either some configuration of your meter is incorrect or the key is somehow invalid. I'm not sure how to help you further... Sorry!
At least I prepared a branch with a fix that the software does not crash anymore. We hope to build soon a new version with this fix.
Sincerely,
raymar9

aselviar, 2022-01-20:
Hi LichtiMC

Do you have the list with the objects that are pushed from your smart meter? If yes, could you share the list?

Another issue may be the mbus adapter you are using, I know of people having experienced problems with those as well. Is it possible to share which adapter you are using and/or try with a different one?

I would also recommend checking again with your utility that the decryption key you enter is valid. 

Best
aselviar

LichtiMC, 2022-01-22:
First of all here you have 10 new lines of raw data for analysis:
```
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 ee 98 db 38 33 c6 78 ff 8e b4 10 fc 8a 2e 25 7b 86 4e ec 3e 6c bd 31 8e af 5d fd 42 b9 1a 48 e7 2a 15 5a 37 f6 8e fa 27 08 73 52 f6 b3 27 14 ea 7a 7a 08 86 1b d2 af 37 8a 6d c4 45 f9 0a 16 55 7d 85 41 99 67 1a d8 ea 4c 55 4a a7 15 05 10 67 95 e5 23 ff ec 6f 7e d4 89 5b 14 6a 05 74 53 c0 36 17 ce db 70 8c 29 b1 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a a0 00 89 bf 58 19 0b 32 5d b2 3a 23 55 4a da 23 ff e0 05 ba 5d 34 3f 65 ce 66 3c c2 e9 be 9e ba e7 8f ac 13 8d 7a 07 a1 f8 0f 82 cc 9a 80 42 ef 5b 88 7e 3f b9 2c 31 ce 27 3f 84 c4 f9 43 33 33 fa eb 66 a3 dc a9 1f 2e 15 06 94 73 7f 7e 12 22 21 a4 c6 2d fd 4a e3 4c ea f2 a9 c7 94 32 b0 aa 07 99 56 95 9a 69 0a 8a a3 aa 51 18 de fe 78 48 81 e2 68 b6 29 18 94 54 87 1d 2a de 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 18 51 22 27 33 a4 1e 5a 4c 31 c3 26 1b 10 51 41 bf 4a ec 92 5b 08 97 6b 84 a0 e0 69 e6 62 00 36 fa 7e
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 ef 9e 37 4e e0 7a d2 d9 79 58 43 8a 7c 36 7c 8a 88 79 5e 6f 1d 55 59 cf 61 a5 6c 52 68 82 51 c7 94 0c 69 be 8d c3 8f 86 de fe 4e 95 3e 1b 72 bf 8f 02 52 76 29 a4 08 ad d0 bf 39 09 ad e6 b0 25 9b 97 86 eb 8f 04 c7 c9 62 4f b8 8c b8 77 79 66 07 6f 90 99 5e ab 2c d3 26 7d 2d 6a 94 e6 7a c3 ed 82 53 f7 8a dc 66 74 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a f1 53 bc c6 d4 b0 fe 8e 8b 90 08 a9 1f ed db f6 77 13 eb b6 19 b9 7f 46 84 b7 d0 b4 50 63 ee 77 79 2a 14 63 42 8f d4 84 9e 61 52 8c 18 1a 95 95 6e 20 a3 60 a5 3c 7b 28 57 53 8c e9 63 55 32 ae b8 e0 27 e2 eb de 04 8b 21 e7 a0 38 52 20 08 79 04 7e af cb 9f 21 50 79 97 46 68 47 91 60 0d d4 ce bc ce 9d 40 7e 67 15 06 bc cd 11 53 4a e2 2d 1c 2a 24 18 44 57 75 3e 5e d0 29 a4 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 97 49 11 eb cd 64 e1 dd f4 c4 4d 29 38 72 3b 96 c7 0f 73 68 5d d7 3a d0 51 c7 a2 38 46 eb 03 53 97 7e
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 f0 4d b4 68 47 e1 a5 e6 59 03 ef 4c 05 cf 52 91 c8 53 26 df 52 65 4e 93 00 9d c3 88 d4 db b4 74 d1 a9 98 1a a0 09 8e 91 51 7b c6 f4 74 1b 96 e1 9e c3 a2 dd 2d 96 85 47 1d fe 0e 2f e6 40 87 62 da c8 bb 14 96 1c a7 70 09 8a 2c c6 94 0a b8 1b 70 e3 56 d4 78 e4 a9 01 64 91 e5 a1 c0 f1 19 a7 8a 7a 47 6f e3 6c 4a e9 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a 6c 4a cf 92 57 53 31 91 d2 40 18 16 e3 9a 34 a6 33 dd 2e 7e fe 3c 68 c9 67 7b 20 2b e2 86 89 2a 58 65 9d 9c d0 32 ba 15 55 09 ee 53 f4 57 0b a4 db d4 d8 3b 22 55 64 5a 4c e4 a7 f1 30 25 04 65 bd 49 f5 9e 43 ad de 02 21 b6 d2 30 14 df 14 34 35 b2 48 88 b1 e7 dc 59 50 f2 d2 f4 16 02 d7 ef 38 bd 17 90 d7 94 84 3a a5 a9 b4 70 30 55 aa 35 8d a2 8f fa 9a bb 08 e9 4b 4c 05 ef 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 23 ee 75 c6 bd 63 50 db ee 20 f6 23 3c 1c 06 6c 21 19 42 10 4a e3 0f ff e7 20 78 26 b7 71 1d 1e 58 7e
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 f1 14 61 9e 55 3e f5 d2 af b0 fb 31 ec 55 ee 8c 80 33 d0 3a 3e ac 6c 74 09 03 5f ab 8f 85 42 f4 30 7c f6 25 57 1d 78 22 c0 62 7f db 55 f0 73 68 c4 56 9f 31 a3 1b 74 37 4c 4d f1 46 d8 59 4a d4 7a 13 2e 36 d5 14 e8 d3 ea 4c 8d 71 c0 9f 27 34 32 a6 9a e0 ab 94 a8 39 1b 83 12 61 16 df 02 f3 a6 bf 7e 46 20 84 17 11 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a 2f 0c 48 ff 7b 16 58 1b 0d 6b 5f 8e 75 62 b1 e3 96 b1 00 0d 94 83 e8 2f 37 32 5b 8c f6 fe 96 67 ba ad b7 02 f2 05 87 89 a2 2d 7e db 80 97 ff 2b 33 1e 8e 3c e2 3d 04 6c 5e 3f f0 a0 e8 c8 85 36 91 b4 30 57 8b 21 38 ac f2 74 e4 67 ef 17 24 70 06 dd 9f 45 d2 a5 7c 25 55 27 3d fc 57 b1 7f e1 0c 46 7e b6 61 60 f1 5c 3e 16 08 5e 21 75 d0 43 e7 8b d4 e2 cf 44 39 ef 33 09 3d 38 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 05 af b9 1f 2c 08 6e 31 5c f9 bb b0 ba 3e e1 0b 5e c9 d5 f2 4c 84 47 6e ce 44 96 b5 a8 4a 14 25 f6 7e
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 f2 8c 40 08 0e aa b4 43 06 32 84 3e e0 e2 26 88 58 e2 c9 0a ce 97 93 b2 8b c8 f7 ff 2e 0f 16 f2 da a6 bd 5d c8 35 51 78 95 b5 51 fb 22 b7 5e 27 4e c4 b0 67 33 85 52 57 72 94 15 68 cd c2 7f de 41 09 35 d3 94 33 7c cf 9c f7 f4 7b 6e 47 6e 1e c2 71 d5 bd 05 48 3b a0 60 d0 51 6d 8b 2a 41 cd c4 12 b7 2d ef 3a 7b 22 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a 30 16 71 0c 1a 47 75 16 d1 ea 41 87 2c 38 f4 67 dc 1c dc 4f 8b f1 57 5c b2 79 2a d6 5c 48 db 31 cb a8 4e d7 8e 48 44 ed ce a8 11 0c df 04 c9 1d 7c 78 23 84 be 54 63 a0 ee 28 3e c7 ab 6c 5d a8 3b d6 43 a4 a5 a8 20 da 15 bd 5f e9 be 71 fe 76 86 e4 1d 8f 9e 3f ec 25 ee cf d7 7b 4b 68 41 3c cf 2b 4a ee e0 93 c6 7d c6 95 4c b1 49 9e 55 d5 9d ee 6c 17 84 92 5b bd 64 39 b8 c9 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 91 56 1b 99 3d f6 04 c6 5d be 86 82 19 b0 b8 19 27 54 41 e9 c5 e5 cc 4c c8 cd e1 6f d8 53 31 9f 1a 7e
7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 78 1f d0 82 01 03 30 00 02 56 f3 bb 49 77 c7 4c 5a f9 9b 94 e1 21 9c 92 f6 5e dd e4 8e a0 40 54 ee b9 7f 48 ce f6 9c 32 2c d1 58 bb 10 a5 4b 31 0b 5d 2d e2 12 f6 60 1a 6b 33 43 bb 4e f4 2e 4d 4a 39 c7 6c ac e0 b9 9f 3a 81 14 05 b8 98 c0 4c 45 3f 81 b9 7d 83 ec d3 98 24 53 96 13 1b d9 9b 7c b2 6b 42 ef 0e c0 14 96 67 c0 eb f5 14 a3 73 b8 66 7e 7e a0 8b ce ff 03 13 ee e1  [... cut ...] 9 f2 15 3a 8e a9 0c 24 fe d8 15 87 56 46 5b e6 f9 f6 bf 2e 2f d0 91 04 2d 38 18 a6 9e 86 24 c2 10 c5 26 3a 66 77 3e 25 e0 98 a3 cd 36 1f a5 d8 c1 d5 2c 96 9f 18 88 59 2e 01 21 88 47 61 eb 36 07 4a 4a 1c e7 b4 be 54 58 b2 2d 18 e9 46 ae 22 d4 e8 b3 24 35 3d 6c 37 2f 9e dd f1 1f 68 25 6c 8f 77 68 da ec ff 98 11 ec 2e 19 3f 7b e3 6f 30 ef d7 f5 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f 61 e7 f2 ea 67 cd 79 35 c0 21 f3 77 d7 e8 29 e5 db 01 2a 16 61 52 43 be ff 1a 7b 66 79 01 7c b2 b1 7e
```

raymar9, 2022-01-25:
I received your E-mail with both keys. I managed to inspect the data with the DLMS Director / Translator Tool https://www.gurux.fi/Download. However, your data seems to be differently structured compared to our L+G E450 meter. And our software (actually the library we use) is unable to parse your data structure.
Actually, it has nothing to do with the authentication key, which is not needed to inspect the data (I tried it with & without the auth key.) Only the encryption key is required. Last time I did not use the correct tool in the DLMS Director to inspect the data.

**Your meter:**

```
<!-- IDIS system title:
Manufacturer Code: LGZ
Device type: IDIS package2 IP polyphase meter
Function type: Disconnector, Load Management, Multi Utility
Serial number: 58204112
 -->
  <DataNotification>
    # Invoke ID: 153327
    <LongInvokeIdAndPriority Value="153327" />
    # 22.01.2022 20:40:10
    <DateTime Value="07E601160614280AFF800000" />
    <NotificationBody>
      <DataValue>
        <Structure Qty="33" >
          # 22.01.2022 20:40:10
          <OctetString Value="07E601160614280AFF800000" />
          # 1.0.1.8.0.255
          <OctetString Value="0100010800FF" />
          <UInt32 Value="4495453" />
          # 1.0.1.8.1.255
          <OctetString Value="0100010801FF" />
          <UInt32 Value="3164981" />
          # 1.0.1.8.2.255
          <OctetString Value="0100010802FF" />
          <UInt32 Value="1330472" />
          # 1.0.1.7.0.255
          <OctetString Value="0100010700FF" />
          <UInt32 Value="1048" />
          # 1.0.2.8.0.255
          <OctetString Value="0100020800FF" />
          <UInt32 Value="0" />
          # 1.0.2.8.1.255
          <OctetString Value="0100020801FF" />
          <UInt32 Value="0" />
          # 1.0.2.8.2.255
          <OctetString Value="0100020802FF" />
          <UInt32 Value="0" />
          # 1.0.2.7.0.255
          <OctetString Value="0100020700FF" />
          <UInt32 Value="0" />
          # 1.0.3.8.0.255
          <OctetString Value="0100030800FF" />
          <UInt32 Value="830134" />
          # 1.0.3.8.1.255
          <OctetString Value="0100030801FF" />
          <UInt32 Value="507549" />
          # 1.0.3.8.2.255
          <OctetString Value="0100030802FF" />
          <UInt32 Value="322585" />
          # 1.0.3.7.0.255
          <OctetString Value="0100030700FF" />
          <UInt32 Value="0" />
          # 1.0.4.8.0.255
          <OctetString Value="0100040800FF" />
          <UInt32 Value="997194" />
          # 1.0.4.8.1.255
          <OctetString Value="0100040801FF" />
          <UInt32 Value="693919" />
          # 1.0.4.8.2.255
          <OctetString Value="0100040802FF" />
          <UInt32 Value="303275" />
          # 1.0.4.7.0.255
          <OctetString Value="0100040700FF" />
          <UInt32 Value="521" />
[... 111 lines cut ...]
              <Int8 Value="2" />
              <UInt16 Value="0" />
            </Structure>
          </Array>
          # 0.9.25.9.0.255
          <OctetString Value="0009190900FF" />
          # LGZ1030653520967
          <OctetString Value="4C475A31303330363533353230393637" />
          # 1909113
          <OctetString Value="31393039313133" />
          # 06.12.2021 01:05:00
          <OctetString Value="07E50C0601010500FF800001" />
          <UInt32 Value="0" />
          <UInt32 Value="134284" />
          <UInt32 Value="0" />
          <UInt32 Value="0" />
          <UInt32 Value="0" />
          <UInt32 Value="0" />
          <UInt32 Value="0" />
          <UInt32 Value="99887" />
        </Structure>
      </DataValue>
    </NotificationBody>
  </DataNotification>
```

LichtiMC, 2022-01-26:
Do you know of a case that works and also uses an L+G E450 and the Energienetze-Steiermark provider?
Is my smart meter malconfigured / defective?
And: Will there be taken some programming effort by you creators of this software to make it work?

LichtiMC, 2022-01-29:
I just saw this fork for my provider:
https://github.com/martink173/smartmeter-datacollector
I will try and ask for help there. Maybe you are able to merge the changes of the fork and make it work for the Energienetze Steiermark provider too!?

raymar9, 2022-01-31:
No, unfortunately I do not know a case where a LG E450 meter from Energienetze Steiermark is used. We only worked with meters from EKZ (Elektrizitätswerke Kantons Zürich).
I suppose it is not defective but differently configured but I am not sure what exactly the problem is.

> Will there be taken some programming effort by you creators of this software to make it work?
Yes, that is for sure. Already analyzing the data, writing tests and such took me a couple hours.

The fork looks promising, however we must analyze it in more detail and certainly adapt a couple of things to be compatible with our software.

LichtiMC, 2022-02-16:
Can you provide a time horizon until when it will work?
Like a week / month / year / etc. ?
Thank you for your efforts!

raymar9, 2022-02-17:
Hi LichtiMC
Yesterday, I tried to integrate the changes from martink173 into our project. The problem is that he actually uses a different implementation for parsing the data from L+G E450 meter model which, as you already know and tried, we provide one by our-self. I run our unittests with his implementation but the tests do not succeed. I suspect that with this modification the data of our meters would not be able to be parsed anymore. We would need to add his parsing implementation separately as a new smart meter type (like L+G E450 Steiermark).
Furthermore, I just posted this problem of the different data structure of our meters in the Gurux forum. Let's see if they know more about this.

Question: Did you try the implementation of martink173 with your meter? Did it work?

Concluding, I cannot guarantee you that we eventually integrate this modification into our project due to the reasons above. Therefore the time horizon would be either a couple of weeks or never. I'm sorry.
If I found a solution for this I would inform you.

LichtiMC, 2022-02-23:
> Question: Did you try the implementation of martink173 with your meter? Did it work?
Yes, it is running on my raspberry pi without a problem.

gcey, 2022-04-14:
I have an L+G E450 from Wiener Netze and I also get the !TypeError: 'int' object is not subscriptable" in "GXDLMSClient.py line 1257".
The "DLMS Director / Translator Tool" decodes to this:

raymar9, 2022-04-20:
I propose that you directly contact Gurux with this request since I won't be able to investigate this issue further. Use this sub-forum to do that: https://www.gurux.fi/forum/25.
If you find out more you can post the link to the forum issue here.

Alternatively, you could use the fork of martink173 (https://github.com/martink173/smartmeter-datacollector).

closed 2022-04-20 by raymar9

gcey, 2022-05-07:
> I propose that you directly contact Gurux with this request since I won't be able to investigate this issue further. Use this sub-forum to do that: https://www.gurux.fi/forum/25. If you find out more you can post the link to the forum issue here.

I did so, see https://www.gurux.fi/comment/23821#comment-23821
There I got the information, that parsePushObjects() is only to be used, if the smart meter sends a list of the objects of the push message as a first parameter.
But the L+G E450 from Wiener Netze does not send this list. It just sends the objects themselves.
In such a case parsePushObjects() must not be called.
Instead the list of objects has to be created with GXDLMSPushSetup.
I found out, that for the L+G E450 from Wiener Netze this can be done in the following way:
```
from gurux_dlms.objects import GXDLMSPushSetup, GXDLMSClock, GXDLMSCaptureObject
...
class HdlcDlmsParser:
    ...
    def __init__(self, cosem_config: Cosem, block_cipher_key: Optional[str] = None) -> None:
        ...  
        self._p = GXDLMSPushSetup()
        self._p.pushObjectList.append((GXDLMSClock(), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.1.1.8.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.1.2.8.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.1.3.8.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.1.4.8.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.0.1.7.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.0.2.7.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.0.3.7.0.255"), GXDLMSCaptureObject(2, 0)))
        self._p.pushObjectList.append((GXDLMSRegister("1.0.4.7.0.255"), GXDLMSCaptureObject(2, 0)))
    ...
    def parse_to_dlms_objects(self) -> Dict[str, GXDLMSObject]:
        parsed_objects: List[Tuple[GXDLMSObject, int]] = []
        if isinstance(self._dlms_data.value, list):
            #pylint: disable=unsubscriptable-object
            parsed_objects = self._p.pushObjectList
            for index, (obj, attr_ind) in enumerate(parsed_objects):
                self._client.updateValue(obj, attr_ind.attributeIndex, self._dlms_data.value[index])
                LOGGER.debug("%d %s %s %s: %s", index, obj.objectType, obj.logicalName, attr_ind.attributeIndex, obj.getValues()[attr_ind.attributeIndex - 1])
        self._dlms_data.clear()
        return {obj.getName(): obj for obj, _ in parsed_objects}
```
IMO this issue should be reopened and the smartmeter-datacollector should be enhanced in the way, that the user can configure, whether
  *  (a) the smart meter sends a list of the objects of the push message as a first parameter (in this case the smartmeter-datacollector should work like ist dies now) or
  *  (b) the smart meter does not send a list of the objects of the push message as a first parameter (in this case the user has to configure, which objects the smart meter sends).

In case of the L+G E450 from Wiener Netze the user has to configure (b) with the following 9 objects:
```
    Clock
    Register 1.1.1.8.0.255
    Register 1.1.2.8.0.255
    Register 1.1.3.8.0.255
    Register 1.1.4.8.0.255
    Register 1.0.1.7.0.255
    Register 1.0.2.7.0.255
    Register 1.0.3.7.0.255
    Register 1.0.4.7.0.255

raymar9, 2022-05-09:
This is some great piece of information. Thank you for your effort! We will try to find a good solution for the integration in smartmeter-datacollector and also [smartmeter-datacollector-configurator](https://github.com/scs/smartmeter-datacollector-configurator) and discuss this extension with EKZ.

I created a new feature-request issue: #31 to further track this extension.

raymar9, 2022-05-09:
@gcey @LichtiMC 
Since we do not possess a L+G meter without the push-list configuration I require some raw data for testing and writing unit tests. Can you provide me with some unencrypted raw data and the corresponding push object list? Or send me some encrypted raw data including the decryption key and the corresponding push object list? However, this unit test code, your data and decryption key is going to be published on this repository, so unencrypted data, if possible, would be the easiest solution.

Please provide that, if you can, in the new issue #31