scs/smartmeter-datacollector #27: frame incomplete and will not be parsed yet --- HDLC byte-buffer > 5000. Buffer is cleared, some data is lost.

gcey, 2022-04-10:
My L+G E450 is from Wiener Netze and I get this (see also [log.txt](https://github.com/scs/smartmeter-datacollector/files/8458629/log.txt)):
Apr 10 10:49:05 localhost systemd[1]: Started Smart Meter Data Collector Service.
Apr 10 10:49:46 localhost smartmeter-datacollector[262]: DEBUG:asyncio:Using selector: EpollSelector
Apr 10 10:49:46 localhost smartmeter-datacollector[262]: INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/ttyUSB0'.
Apr 10 10:49:55 localhost smartmeter-datacollector[262]: DEBUG:smartmeter:HDLC Buffer: 7B EE 8F 54 1F 01 00 8B D5 13 13 A1 C5 09 68 20 3B 07 6D DA 24 01 35 1F F9 7B EE 8F D4 07 01 11 B2 0A AF 7B 73 1B 61 43 17 02 49 02 D8 0F 40 E2 36 FD 7B EE 8F 54 1F 01 61 8D FC 66 F1 81 37 BB A3 37 5F 17 3A 5C 77 18 17 F3 7B EE 87 50 0F 01 31 89 44 D4 4B 4A 67 0A 73 56 33 19 32 E4 5B 1F F7 0D FE 7B EE 8F D4 1F 01 71 0C BF E1 19 29 68 DE 47 D0 09 A2 81 6F 68 95 0E AD FE 03 EE 87 50 1F 01 A3 0C 1A 0A C8 C6 E0 5A 9B D1 09 08 07 C1 3E 07 9B B0 7B EE 8F D4 1F 01 73 2E 3F 57 0A D4 20 76 43 E6 05 FD 6C 1F 1F 03 4D 6A 7B AE 8F D4 1F 01 F3 E1 E9 1B 5B 0A 7C C0 64 0E E8 0C CD 7D 20 49 7C FF 3B EE 8F 50 1F 01 03 03 E6 F0 EE B5 5A A7 3B 0A 1E 06 1E 53 E4 A3 1A FF 7B 8E 8F D4 1F 01 01 DD 7E
Apr 10 10:49:55 localhost smartmeter-datacollector[262]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
...
Apr 10 10:53:01 localhost smartmeter-datacollector[262]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Apr 10 10:53:19 localhost smartmeter-datacollector[262]: DEBUG:smartmeter:HDLC Buffer: 7B EE 8F 54 1F 01 00 8B D5 13 13 A1 C5 09 68 20 3B 07 6D DA 24 01 35 1F F9 7B EE 8F D4 07 01 11 B2 0A AF 7B 73 1B 61 43 17 02 49 02 D8 0F 40 E2 36 FD 7B EE 8F 54 1F 01 61 8D FC 66 F1 81 37 BB A3 37 5F 17 3A 5C 77 18 17 F3 7B EE 87 50 0F 01 31 89 44 D4 4B 4A 67 0A 73 56 33 19 32 E4 5B 1F F7 0D FE 7B EE 8F D4 1F 01 71 0C BF E1 19 29 68 DE 47 D0 09 A2 81 6F 68 95 0E AD FE 03 EE 87 50 1F 01 A3 0C 1A 0A C8 C6 E0 5A 9B D1 09 08 07 C1 3E 07 9B B0 7B EE 8F D4 1F 01 73 2E 3F 57 0A D4 20 76 43 E6 05 FD 6C 1F 1F 03 4D 6A 7B AE 8F D4 1F 01 F3 E1 E9 1B 5B 0A 7C C0 64 0E E8 0C CD 7D 20 49 7C FF 3B EE 8F 50 1F 01 03 03 E6 F0 EE B5 5A A7 3B 0A 1E 06 1E 53 E4 A3 1A FF 7B 8E 8F D4 1F 01 01 DD 7E F5 73 02 3E 40 07 70 26 FC 06 7C 3A 07 EC 17 81 03 EE 8F 50 1F 01 31 9F 7F 53 9D 45 1A 1B 36 52 0B 7E 03 6C CD 8D 17 5F FF 7B 8A 8F D4 1F 01 51 56 3A 5E 56 60 15 6C 6F 40 6A 7E 1F 4A C8 7E 8A 75 FE 7B 8A 8F D4 1F 01 A1 33 16 5F 81 B3 4D F2 36 1A 52 77 12 75 62 53 A3 7B EE 8F 50 1F 00 13 13 4A 1F AF 01 0D B8 01 8D 51 62 72 39 67 09 FB C9 4B 7B EE 8F 50 1F 01 53 60 67 72 16 C8 15 12 63 3F 5A 4C E4 23 08 3E 2C FF 03 EE 8F 50 1F 01 B3 63 9B 57 70 91 1C 17 5D F1 D9 24 01 6B 23 7B 09 78 03 EE 8F D4 1F 01 C3 01 19 37 09 8F 0A E0 23 3E 07 17 67 1E B1 5C 73 24 85 7B EE 87 50 1F 01 41 5A 7E 02 36 27 B3 2E 57 7B 23 3A 76 77 17 5A 78 26 FE 03 EE 8F 54 1F 01 21 88 E0 99 E0 6C 37 D0 21 58 0B AD D1 0F 36 C5 F1 79 07 FF 7B EE 8F 50 1F 01 51 7A 49 73 02 76 2C E2 22 13 6F 68 5F 5F A6 52 02 F4 83 7B 8A 8F 50 1F 01 F1 69 40 CE 1A 29 AC 37 08 5D F1 1B 16 85 5F 1E 7F 73 FF 7B 8E 8F D4 05 01 63 A9 4C F8 3A F2 0F 7A 03 36 25 20 6F 26 E0 BF 60 3A 23 1B EE 8F 54 1F 01 23 AD 70 1B 07 41 09 84 0E 7F 49 09 D1 FF CC 13 07 1F 77 FF 7B EE 87 50 0F 01 03 8A 1A 5B 5A E0 33 03 59 3B 16 15 2A 6C E5 28 41 F7 7B 8E 8F 54 1F 01 03 2F CE 1B 1F 57 07 E3 09 03 B6 77 54 06 7C 09 A5 FF 7B EE 8F 54 1F 01 F1 36 3E D5 B6 17 E8 02 F8 17 D4 4F 13 57 73 50 7D 49 FA 7B EE 8F D4 1F 01 61 E4 E5 5B 27 40 79 32 17 75 B8 44 09 73 F4 8A D8 7A 5E 7B EE 8F 54 1F 01 E1 9F 32 3A 60 AC 0F 65 16 7E 12 1B 37 33 3F 12 E9 A2 7B 8A 8F D4 1F 01 31 59 B5 03 17 13 1A 12 1E 13 26 16 F7 F4 A2 49 72 3A FF 7B EE 8F 50 1F 01 43 E5 1E 46 34 03 E8 CF F0 50 0A 44 5E 1F D1 5A 07 5F FE 7B EE 87 50 1F 01 53 57 4A 7F 13 3B 3B 02 1F AC F3 B4 BF 47 57 76 68 77 F5 7B EE 8F 50 1F 01 E0 73 7B 27 ED 1F 7A 73 57 1B 1D 6E 40 98 9C 3F 57 7D FE 7B EE 8F 50 1F 01 73 07 8B 19 EC 02 1B 1F 8D C9 42 35 FC F8 48 E6 1F 1B FF 7B EE 8F 50 1F 01 11 C5 0B 7A 33 7D FC 40 C7 03 1F C8 2A 3D 7C 19 40 50 6B F5 3B EE 8F D4 1F 01 A0 C1 B9 1E 0C A9 33 56 A6 38 EE BB 23 09 CC FC 16 3B 5E 7B AE 8F D4 05 01 01 A5 AB DB 5C 2B 5F 36 72 40 3F 45 0F 41 37 13 12 73 FF 1B EE 8F 50 1F 01 51 8F 5E 2C 2F 2A A3 3B DB 08 3B C8 43 53 06 3B 0C 00 D1 7B EE 8F 50 1F 01 53 0F 63 7B 93 2C 9B BD CE C4 22 BB 7E 59 03 12 63 57 12 F4 7B EE 8F 50 1F 01 13 3B 48 89 1B DC 37 F5 11 BB B2 1E 1F 8F 27 EE 04 64 FF 03 EE 8F 50 1F 01 23 E5 FC 31 08 F4 1F A1 60 05 62 E6 3F 3F A2 62 1F EA E0 03 EE 8F D4 1F 01 83 6A 72 40 2A 3B 12 1E 33 1B 48 53 2A 7B 22 7A 70 A9 FF 7B 8A 8F D4 05 01 A1 08 59 6E 1A BB 64 22 E8 77 11 17 33 89 5B 1F 5A FB 7B 8A 8F 50 1F 01 D1 07 32 02 37 14 1E 7A 50 DE E6 22 37 42 17 E8 26 3E 3B FF 7B EE 8F 50 0F 01 11 D0 45 73 5B 48 0E 07 56 42 37 03 13 06 33 22 5E 52 E0 7B EE 8F 50 1F 01 F1 A9 A1 18 C9 06 F8 02 4C 1A 5A 37 36 7E 56 7C A9 2B 58 FA 7B 8E 8F D4 1F 01 23 A6 6C 87 09 AD 5E 12 73 17 3A 1E 17 12 22 26 35 41 78 7B EE 8F 50 1F 00 43 22 7B C4 63 62 57 2C D3 08 5A 22 05 A1 98 CA C1 33 FA 03 EE 8F 50 1F 01 F3 18 EF 4B 2B 75 3F D9 DC 68 7B 73 B4 C0 C5 30 A6 0D 7B EE 8F D4 1F 00 63 60 75 56 13 72 32 02 1A 57 5B 57 7E 5A C8 0E 16 7F 0F 7B EE 8F 54 1F 01 E1 3F 39 9F 48 42 AC 22 C1 3A 1F 73 5B 73 1B CC 32 17 FA 7B AE 8F D4 1F 01 91 16 53 22 77 53 7A 16 7D F0 FD 77 17 36 67 32 24 58 FF 7B EE 8F 50 1F 01 21 11 6F 56 76 07 1E 36 7B 8E 48 41 03 3E 75 AC 75 13 FE 7B EE 8F 50 1F 01 61 A9 AF FE AD 0E 1B 72 26 73 E8 84 3D 15 81 EA AB 71 43 FE 0B EE 8F 50 1F 01 E3 7B 73 0E 2D 2B 20 E8 31 78 39 5C 72 5E 33 7F 13 FE 7B EE 8F 54 1F 01 23 13 1B 2E 99 20 88 3E 12 57 0B 5A 46 36 56 F4 0C F3 FF 7B EE 8F D4 07 01 33 B8 19 ED FD 49 02 6A 1B 5F 2C 76 E9 F5 72 3E 7E FE 03 EE 8F D4 0F 01 03 67 0F 00 24 BD 2F F9 17 48 C2 33 6E 78 4B B5 17 0A FF 7B 8E 8F D4 05 01 95 75 40 C6 75 C2 23 3F 3F 22 3B 8D 97 2A 1B 1A 01 F9 7B 8A 8F 54 1F 01 C5 B8 1B F1 81 3E 3E 02 7F 59 03 5A C8 73 52 02 37 41 FB 7B EE 8F 50 1F 01 55 74 89 7D 36 33 3B 5F 5B 4D F5 1B 75 3A DD F8 67 77 FE 03 EE 8F 50 1F 01 35 2B 45 E4 C1 1A 21 D5 DE B8 49 09 4A 3F 26 03 D0 AA F9 7B EE 8F 50 1F 01 67 57 3A 0E 5E 2E 56 05 5D ED 03 7E 7B 57 56 B5 3F 7B EE 8F 50 1F 01 B7 1B 1A 26 57 4A 18 4C 7A 20 24 47 1B 7D F1 73 02 37 FB 7B 8A 8F 54 1F 01 77 2A 66 D3 99 F0 43 33 3B 0A EC A8 47 47 56 25 17 23 7B EE 8F 50 1F 01 C7 36 E6 25 E7 AF 3F D0 E8 CC 1A 77 3A 6A 77 12 CF EF 85 7B 8A 8F 50 1F 01 05 DC 0E 37 53 76 73 03 17 41 D5 38 91 3A A9 5A 89 3B EC 03 EE 8F D4 07 01 75 E9 32 33 57 FA 13 4E 5F 27 F5 6A 81 7F 7E 2D 66 B6 1B EE 8F D4 1F 01 15 A8 4F 83 19 45 D5 78 A1 07 3A 05 17 16 3B CF 39 6F FE 7B EE 8F 50 1F 01 F5 0F DE 19 50 83 0D E5 7A 36 5E 5B A3 41 A7 76 60 05 1F 7B EE 8F D4 05 01 D7 72 44 71 1E 72 4B 57 0D 93 17 48 FF AB 06 65 5F 57 32 7B EE 8F 50 1F 01 97 17 A0 33 7F E5 1D 15 BF 27 29 12 76 F3 0B 41 E7 8C D5 7B 8E 8F D4 1F 01 A7 2D 45 07 37 28 A0 BF C1 0A 16 EC 4A 7B 76 12 19 55 27 FF 7B 8E 8F 54 1F 01 A7 34 78 1D AC 6D 7F 68 88 52 02 EC A7 2D 33 88 71 02 FF 7B EE 8F D4 05 01 55 25 A5 24 01 CB A2 16 0D D8 03 C9 53 1B 73 06 7D C8 03 EE 8F 54 1F 01 15 89 70 23 3E 3A EA 63 5E 46 06 1F 5F F1 7F CC E5 FE FF 03 EE 87 50 0F 01 61 09 01 5E 44 5E 0C 1B C5 EE 6E 8C 66 70 5F 29 4C 3B FE 03 EE 8F 50 1F 01 F5 3C 2D 63 19 1F 37 1E 32 3A CC 4C 30 0F 44 61 42 42 37 7B 8E 8F 54 1F 01 57 66 07 45 5E 33 77 51 C3 0F 03 1D 13 03 19 BC 02 5A FF 7B EE 8F D4 07 01 27 0C 71 0B FF F8 BD B0 77 33 77 73 17 AB 16 06 FA 6B 7B EE 8F D4 1F 01 E7 A3 DB 48 5B 06 7D 3B 8F E4 B9 68 D4 2B 03 73 1E E1 FE 7B EE 8F D4 1F 01 67 51 3A 2E 73 E3 FA 0D 65 13 7B 26 AF C0 73 26 35 FB 1B EE 8F 50 1F 01 65 5A 0C 41 A3 64 26 0B 59 68 02 44 03 1B 09 FE 0E 5F 87 0B EE 8F 50 1F 01 55 3C 5C BD 32 D3 11 ED A0 0D 53 1B 62 5B 55 0F FC 1A F7 7B EE 8F 54 1F 01 05 73 5E 6F 0B E1 29 8B C8 75 0F 3F 36 1F 7A 36 7D 03 2E FF 7B EE 87 58 1F 01 30 09 05 BD 26 17 73 49 A7 2E E9 CC 7C 0E 55 56 63 C9 27 03 EE 8F 54 1F 01 37 7E 0E 3A 37 55 66 E6 32 01 2E 02 5F 0B 3F 75 3E 99 FF 7B 8E 8F 54 1F 01 02 A6 35 36 B9 08 8B 23 5C 09 AF B5 83 48 AD 45 C5 36 1A 7B AE 8F 54 1F 01 37 20 53 C1 52 D3 2B 17 7F 3F 5F 86 64 0A 5B 43 3A A1 7B 8A 8F 50 1F 01 27 22 5A D5 4A BD 60 98 B4 E2 22 55 1A 1F 7D 12 CC 7E E8 0B EE 8F 50 1F 01 25 28 43 63 16 02 B6 0E 7F 2D 72 0F B8 21 D7 3D 02 05 FC 7B 8E 8F 54 1F 00 D5 0B 16 07 7B 24 FD 22 36 1B 2E EF 53 C9 C4 02 57 77 E8 7B EE 8F D4 1F 01 D5 47 44 27 7A 1E 50 43 3E DC 86 33 2E 16 02 76 9F 61 91 7B 8A 8F 54 1F 01 65 6A 0B C8 73 0B 13 0E 17 58 17 24 8B 77 3F 77 1A 8D C8 1B  [... cut ...]  01 05 3A 1D 48 AE AD C8 06 72 BC 9D C9 06 5E 7A 72 D4 73 EC 7B EE 8F D4 1F 01 25 57 5B 2A 73 0B 69 03 46 73 E4 23 36 13 E4 72 1F 76 E0 03 EE 8F 54 1F 01 E5 23 48 44 B2 BF C1 17 AE 14 70 64 AC 17 1A 12 77 D9 22 7B 8A 8F D4 05 01 47 7A 7E
Apr 10 10:53:20 localhost smartmeter-datacollector[262]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Apr 10 10:53:20 localhost smartmeter-datacollector[262]: WARNING:smartmeter:HDLC byte-buffer > 5000. Buffer is cleared, some data is lost.

raymar9, 2022-04-11:
Hi,
this looks like that no valid or decodable HDLC frame could be detected in the data sent by the meter.
`HDLC byte-buffer > 5000` is deliberate and avoids filling up memory indefinitely if invalid data is gathered by the software due to some reason.
Unfortunately, I cannot help you further. Maybe you have a incompatible smart meter configuration, e.g. not using HDLC / DLMS.

gcey, 2022-04-11:
If I send you my decryption key, could you then try to find out, what is wrong?
Would be great!

raymar9, 2022-04-11:
No, I'm sorry, this won't help. I cannot detect any valid HDLC frames in your data, therefore the decryption key would not help on. Please check that your meter is sending valid HDLC frames over the wired MBus serial dongle. Is the Baudrate correct? Do you use wired MBus and HDLC / DLMS with your meter?

You can use the Gurux DLMS Director (DLMS Translator Tool), https://www.gurux.fi/GXDLMSDirector for decoding the received data. If this tool cannot parse the data, ours cannot neither.

You could also post your questions concerning HDLC, DLMS, ... to the Gurux forum (https://www.gurux.fi/forum).

gcey, 2022-04-18:
I found out, that the assumption in smartmeter-datacollector is wrong, that each L+G E450 meter communicates with 2400 bps.
My L+G E450 meter from Wiener Netze communicates with 9600 bps on the optical IR port.
I will write a separate issue for this.
When I modify smartmeter-datacollector to use 9600 bps, then I get valid HDLC frames.
The DLMS data in these frames can be decoded by the Gurux DLMS Director (DLMS Translator Tool), https://www.gurux.fi/GXDLMSDirector - but smartmeter-datacollector fails in decoding these data.
Do these two tools use different DLMS decoders?

raymar9, 2022-04-20:
I do not know about the internal implementation details in the Gurux project, how they maintain the different code bases in the supported languages. However, the GXDLMS Director and its tools are probably written in C#/.net and the smartmeter-datacollector uses their Python implementation. Further, I suspect that the DLMS translator is more open to various data structures compared to our software since the translator is a debugging tool and the parsed data only needs to be displayed.

closed 2022-04-20 by raymar9