scs/smartmeter-datacollector #30: Get water meter data from l+g e450

dawiinci, 2022-05-02:
Is the water meter accessible with this repository? As far as I understand, the e450 collects and transmits this data as well.

raymar9, 2022-05-05:
No, only the OBIS codes noted [here](https://github.com/scs/smartmeter-datacollector/blob/master/smartmeter_datacollector/smartmeter/lge450.py) are parsed by the datacollector. You would have to extend this list and add new MeterDataPointTypes [here](https://github.com/scs/smartmeter-datacollector/blob/d4613f504b7b0e1bb63af14c67f200402a6db744/smartmeter_datacollector/smartmeter/meter_data.py).

closed 2022-05-05 by raymar9

dawiinci, 2022-05-05:
Thanks, I guess I can do that. 

I have some questions, though.

This is an example of what I have to add to both .py

```
RegisterCosem("1.0.4.7.0.255", MeterDataPointTypes. WATER_FLOW.value),

WATER_FLOW = MeterDataPointType("WATER_FLOW", "Water Flow", "l/s")
```

How do I find the address of this fictional "WATER_FLOW" and probably a lot of other available data points? Is there some documentation/Standard or is this known by the EVU or the manufacturer?

As far as I understand, this is the only part that has to match the device, the rest is internal variables for the sinks.

raymar9, 2022-05-05:
Look for OBIS code lists on the web like this one: https://www.promotic.eu/en/pmdoc/Subsystems/Comm/PmDrivers/IEC62056_OBIS.htm
Since you probably need to configure your smart-meter such that it pushes out the data you want through its M-BUS interface, you can also find the correct OBIS codes in the config tool of the meter. 
If you want to manually record and analyze data from your meter I suggest using [Gurux DLMS Director](https://www.gurux.fi/GXDLMSDirector). It provides two relevant tools: Serial Monitor & DLMS Translator. Record data with the Serial Monitor and parse it with the DLMS Translator.

dawiinci, 2022-05-05:
That makes sense. So if I get the config tool, I will have the correct OBIS codes and I can add it to python scripts. I only need the values in MQTT. 

Since I anyway have to enable it, I need that config tool. However, I have no idea where to get that for my meter.

raymar9, 2022-05-05:
We developed the smartmeter-datacollector for the electricity supplier EKZ, who is the only one with access to the officially installed smartmeters the buildings. If you have an official meter, your electricity provider must enable and configure this data interface. I highly doubt that you have access yourself, even if you get hold of this software tool.
Unfortunately, I can't help you further since I haven't configured a smart-meter myself yet.

aselviar, 2022-05-11:
@dawiinci : Thank you for your question. Regarding the meter values, it first of all depends if the e450 is connected via MBus with the water meter, which is something that your electricity/water provider should know. If this is the case, e450 can push these values via the local interface in one of the four registers (namely the fourth register named "Consumer Information 4"), but only if configured so by your provider. Please ask your provider about what is being pushed from the local interface and what are the OBIS codes you should look for regarding the water meter (code for M-Bus device ID from channel x, M-Bus value from channel x).

Hope this hels!

kafisatz, 2022-08-24:
@dawiinci did make any progress on this? My water meter is connected to my E450 and I am trying to extract the data too.

dawiinci, 2022-08-24:
There is no live data from the water meter. Only maybe once a day or so. 

This would be possible but requires a lot of work from the distributor.

kafisatz, 2022-08-25:
Thanks. As an alternative, I have been thinking about directly accessing the water meter. If you also have a GWF water (https://i.imgur.com/eQrD9Q1.jpg) can you possibly measure the voltage on its connection? I measure almost zero voltage on mine, thus it does not seem to be an mbus connection (which should have some 25-36V).

dawiinci, 2022-08-26:
I have the same. I think it is only active when the master meter reads. 

One possibility I have seen is to use a camera to get live data using image recognition.

kafisatz, 2022-09-26:
@dawiinci I have no successfully read data from the GWF water meter. However, I need to **disconnect** the electricity meter, thus my energy provider will not get any information about water consumption anymore, which they won't like even though they only need the value once per year (as far as I know). I was using a USB to MBus **Master** device to read out the data (on Windows 11). With a slave MBus device I was not able to read any data (irrespective if the GWF water meter was connected to the electricity meter or not). I think with an [MBus splitter (overkill...)](https://shop.marcomweb.it/en/shop-online/fieldbus/meterbus-m-bus/m-bus-splitter-multimaster-for-4-meter-dettagli.html) I could make this work while having the electricity meter connected.