scs/smartmeter-datacollector #33: Iskraemeco AM550 - Romande Energie

xens, 2022-08-09:
I've been struggling to read the data from my new Iskraemeco AM550 smartmeter from my electricity provider Romande Energie.

The smartmetter is connected to a raspi using a P1 cable, the config for smartmeter-datacollector is the following:

```ini
[reader0]
type = iskraam550
port = /dev/ttyUSB0

[sink0]
type = logger
name = DataLogger

[logging]
default = DEBUG
smartmeter = DEBUG
collector = DEBUG
sink = DEBUG
```
Here's the data I obtain when running smartmeter-datacollector in DEBUG mode

```
smartmeter-datacollector -c config.ini 
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up Iskra AM550 smart meter on '/dev/ttyUSB0'.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 16 1E 02 09 01 01 08 FF 00 47 02 0F 16 2C 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 16 1E 02 09 01 01 08 FF 00 47 02 0F 16 2C 7E 7E CF 02 03 96 E7 0F 1A D1 07 08 02 07 00 88 02 02 09 00 60 00 09 37 34 34 30 38 02 06 00 01 FF 06 32 33 33 03 06 01 01 07 FF 00 01 02 0F 16 02 09 01 02 00 06 00 00 02 0F 16 02 09 01 01 00 06 15 5C 02 00 1E 03 06 01 08 FF 00 E3 02 0F 16 02 09 01 20 00 64 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 16 1E 02 09 01 01 08 FF 00 47 02 0F 16 2C 7E 7E CF 02 03 96 E7 0F 1A D1 07 08 02 07 00 88 02 02 09 00 60 00 09 37 34 34 30 38 02 06 00 01 FF 06 32 33 33 03 06 01 01 07 FF 00 01 02 0F 16 02 09 01 02 00 06 00 00 02 0F 16 02 09 01 01 00 06 15 5C 02 00 1E 03 06 01 08 FF 00 E3 02 0F 16 02 09 01 20 00 64 7E A8 A4 CF 02 23 03 99 96 FF 12 09 55 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 FF 61 02 FF 23 03 06 00 07 FF 09 02 0F 16 02 09 01 1F 00 12 99 02 FE 21 03 06 00 07 FF 00 02 0F 16 02 09 01 47 00 12 3D 02 FE 21 03 06 01 08 FF 00 34 02 0F 16 02 09 01 01 02 06 0B E9 02 00 1E 03 06 01 F3 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 16 1E 02 09 01 01 08 FF 00 47 02 0F 16 2C 7E 7E CF 02 03 96 E7 0F 1A D1 07 08 02 07 00 88 02 02 09 00 60 00 09 37 34 34 30 38 02 06 00 01 FF 06 32 33 33 03 06 01 01 07 FF 00 01 02 0F 16 02 09 01 02 00 06 00 00 02 0F 16 02 09 01 01 00 06 15 5C 02 00 1E 03 06 01 08 FF 00 E3 02 0F 16 02 09 01 20 00 64 7E A8 A4 CF 02 23 03 99 96 FF 12 09 55 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 FF 61 02 FF 23 03 06 00 07 FF 09 02 0F 16 02 09 01 1F 00 12 99 02 FE 21 03 06 00 07 FF 00 02 0F 16 02 09 01 47 00 12 3D 02 FE 21 03 06 01 08 FF 00 34 02 0F 16 02 09 01 01 02 06 0B E9 02 00 1E 03 06 01 F3 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 02 FF 06 00 06 47 02 02 00 1E 69 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E 07 00 88 02 02 09 00 60 00 09 37 34 37 30 02 09 00 60 01 09 31 39 33 02 09 01 01 00 06 00 5B 02 00 1B 03 06 01 07 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 0F 16 02 09 01 02 00 06 13 81 02 00 1E 03 06 00 07 58 7E CF 23 99 FF 09 02 0F 16 02 09 01 34 00 12 62 02 FF 02 09 01 48 00 12 64 02 FF 23 03 06 00 07 FF 00 02 0F 16 02 09 01 33 00 12 10 02 16 02 09 00 07 FF 00 02 0F 16 02 09 01 01 01 06 0A 73 02 00 1E 03 06 01 08 FF 00 42 02 0F 1E 03 06 01 B8 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 16 1E 02 09 01 01 08 FF 00 47 02 0F 16 2C 7E 7E CF 02 03 96 E7 0F 1A D1 07 08 02 07 00 88 02 02 09 00 60 00 09 37 34 34 30 38 02 06 00 01 FF 06 32 33 33 03 06 01 01 07 FF 00 01 02 0F 16 02 09 01 02 00 06 00 00 02 0F 16 02 09 01 01 00 06 15 5C 02 00 1E 03 06 01 08 FF 00 E3 02 0F 16 02 09 01 20 00 64 7E A8 A4 CF 02 23 03 99 96 FF 12 09 55 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 FF 61 02 FF 23 03 06 00 07 FF 09 02 0F 16 02 09 01 1F 00 12 99 02 FE 21 03 06 00 07 FF 00 02 0F 16 02 09 01 47 00 12 3D 02 FE 21 03 06 01 08 FF 00 34 02 0F 16 02 09 01 01 02 06 0B E9 02 00 1E 03 06 01 F3 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0D 9C 2B 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 02 FF 06 00 06 47 02 02 00 1E 69 7E FF 08 38 34 30 38 02 06 00 01 FF 06 32 33 33 03 06 01 07 FF 00 6A 02 00 1B 03 06 01 07 00 06 00 00 02 00 1B 03 06 01 08 FF 00 77 02 02 00 1E 03 06 01 08 FF 00 E3 02 0F 16 02 09 01 20 00 2C 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
```

According to Romande Energie documentation, here's the technical specs of the smartmeter (see original PDF attached in french):

## Device settings
• Connector: RJ12 on the top right corner of the device, labelled « interface P1»
• Interface: P1 DSMR V5.0
• Protocol: DLMS/COSEM
• Frame lenght: 154 bytes

## Device configuration
• 115200bds
• 8 bits
• Parity - None
• 1 stop bit

[Spécifications techniques ISKRA.pdf](https://github.com/scs/smartmeter-datacollector/files/9293944/Specifications.techniques.ISKRA.pdf)

After exchanging some email with the Romande Energie support, they validated that the data on the P1 port is not encrypted.

do you have any ideas or hints to make me progress with this issue ?

raymar9, 2022-08-16:
Hi 
Unfortunately, I couldn't decode the sample data you provided, too. I tried it with the [DLMS translator](http://www.gurux.fi/gurux_dlms_translator45/Gurux_dlms_translator.html). The result was just empty, no error, no warning.

Did you run the software for some time until the buffer was cleared?
Can you provide some more example data (log) here that I can have another go with the DLMS-translator?
The serial settings seem to be correct and correspond to those in the software.

You could also install the [Gurux DLMS Director](http://www.gurux.fi/Download) and use the tools "Serial Monitor" and "DLMS Translator" to receive serial data from the dongle on your computer and parse it then with the translator.

xens, 2022-08-16:
@raymar9, I've followed your advice and used the Gurux DLMS Directory, extracted data using the serial monitor (see [attached file](https://github.com/scs/smartmeter-datacollector/files/9352593/serial_out.txt)) and parsed it using the DLMS Translator.

Here's the output for one PDU

```xml
BlockCipher key: 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
Authentication Key:D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF
1: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 0F 00 1C 37 29 0C 07 E6 08 10 02 11 34 23 00 FF 88 80 02 10 02 02 09 06 00 00 60 01 00 FF 09 08 37 38 34 34 37 30 30 38 02 02 09 06 00 00 60 01 01 FF 09 06 31 32 39 33 33 33 02 03 09 06 01 01 01 07 00 FF 06 00 00 00 00 02 02 0F 00 16 1B 02 03 09 06 01 01 02 07 00 FF 06 00 00 01 46 02 02 0F 00 16 1B 02 03 09 06 01 01 01 08 00 FF 06 00 16 1A 4E 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 00 FF 06 00 14 99 A7 02 02 0F 00 16 1E 02 03 09 06 01 00 20 07 00 03 F2 7E 7E A8 A4 CF 02 23 03 99 96 FF 12 09 88 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 00 FF 12 09 7B 02 02 0F FF 16 23 02 03 09 06 01 00 48 07 00 FF 12 09 7D 02 02 0F FF 16 23 02 03 09 06 01 00 1F 07 00 FF 12 01 7E 02 02 0F FE 16 21 02 03 09 06 01 00 33 07 00 FF 12 00 BF 02 02 0F FE 16 21 02 03 09 06 01 00 47 07 00 FF 12 00 38 02 02 0F FE 16 21 02 03 09 06 01 01 01 08 01 FF 06 00 0A 6A 59 02 02 0F 00 16 1E 02 03 09 06 01 01 01 08 02 FF 06 00 0B AF F5 02 02 0F 00 16 1E 02 03 09 06 01 01 02 25 28 7E 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0E 29 AC 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 02 FF 06 00 06 6F FB 02 02 0F 00 16 1E 11 5D 7E
<HDLC len="178" >
<TargetAddress Value="67" />
<!-- Logical address:1, Physical address:17 -->
<SourceAddress Value="91" />
<!-- Notification frame. -->
<FrameType Value="13" />
<PDU>
<DataNotification>
  <!-- Invoke ID: 1849129 -->
  <LongInvokeIdAndPriority Value="1849129" />
  <!-- 8/16/2022 8:52:35 AM -->
  <DateTime Value="07E608100211342300FF8880" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="16" >
        <Structure Qty="2" >
          <!-- 0.0.96.1.0.255 -->
          <OctetString Value="0000600100FF" />
          <!-- 78447008 -->
          <OctetString Value="3738343437303038" />
        </Structure>
        <Structure Qty="2" >
          <!-- 0.0.96.1.1.255 -->
          <OctetString Value="0000600101FF" />
          <!-- 129333 -->
          <OctetString Value="313239333333" />
        </Structure>
        <Structure Qty="3" >
          <!-- 1.1.1.7.0.255 -->
          <OctetString Value="0101010700FF" />
          <UInt32 Value="0" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="27" />
          </Structure>
        </Structure>
        <Structure Qty="3" >
          <!-- 1.1.2.7.0.255 -->
          <OctetString Value="0101020700FF" />
          <UInt32 Value="326" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="27" />
          </Structure>
        </Structure>
        <Structure Qty="3" >
          <!-- 1.1.1.8.0.255 -->
          <OctetString Value="0101010800FF" />
          <UInt32 Value="1448526" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="30" />
          </Structure>
        </Structure>
        <Structure Qty="3" >
          <!-- 1.1.2.8.0.255 -->
          <OctetString Value="0101020800FF" />
          <UInt32 Value="1350055" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="30" />
          </Structure>
[... 75 lines cut ...]
          <OctetString Value="0101020801FF" />
          <UInt32 Value="928172" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="30" />
          </Structure>
        </Structure>
        <Structure Qty="3" >
          <!-- 1.1.2.8.2.255 -->
          <OctetString Value="0101020802FF" />
          <UInt32 Value="421883" />
          <Structure Qty="2" >
            <Int8 Value="0" />
            <Enum Value="30" />
          </Structure>
        </Structure>
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
</PDU>
</HDLC>
```

I guess this is a good sign as I can ready the data right ?

raymar9, 2022-08-18:
Yes, that looks good. Can you try it again with the smartmeter-datacollector and record some more log messages with raw data?

I do not know why the data seems to be invalid when using the datacollector (no valid HDLC frames). All HDLC frames should either start with 7E A0 or 7E A8 which the serial monitor recorded data does and the data logged by the datacollector doesn't.
Sometimes, this happens if the wrong serial settings are applied, however I double-checked yours with the default config of the AM550 in our code.

xens, 2022-08-18:
Ok I really don't understand why but now it's going further since I've read the data from the DLMS Director, previously on different machines I always had the error from my initial post. I'm wondering if I messed up my smartmeter at some point and DLMS Directory reseted it using some magic code... or maybe I've got something wrong going on somewhere else.

Anyways it's now going further and now crashing on "Invalid OBIS Code".

If I copy/paste this frame on the DLMS Director I don't receive any error message and it seems to be happy with it. Any idea ? 

```
smartmeter-datacollector -c config.ini -d
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up Iskra AM550 smart meter on '/dev/ttyUSB0'.
DEBUG:smartmeter:HDLC Buffer: 00 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
INFO:asyncio:poll took 2035.229 ms: 1 events
DEBUG:asyncio:poll took 15.872 ms: 1 events
DEBUG:smartmeter:HDLC Buffer: 00 7E 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 0F 00 1C 38 10 0C 07 E6 08 12 04 15 1B 32 00 FF 88 80 02 10 02 02 09 06 00 00 60 01 00 FF 09 08 37 38 34 34 37 30 30 38 02 02 09 06 00 00 60 01 01 FF 09 06 31 32 39 33 33 33 02 03 09 06 01 01 01 07 00 FF 06 00 00 01 CA 02 02 0F 00 16 1B 02 03 09 06 01 01 02 07 00 FF 06 00 00 00 00 02 02 0F 00 16 1B 02 03 09 06 01 01 01 08 00 FF 06 00 16 79 50 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 00 FF 06 00 14 B9 D9 02 02 0F 00 16 1E 02 03 09 06 01 00 20 07 00 14 3A 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:asyncio:poll took 16.033 ms: 1 events
DEBUG:smartmeter:HDLC Buffer: 00 7E 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 0F 00 1C 38 10 0C 07 E6 08 12 04 15 1B 32 00 FF 88 80 02 10 02 02 09 06 00 00 60 01 00 FF 09 08 37 38 34 34 37 30 30 38 02 02 09 06 00 00 60 01 01 FF 09 06 31 32 39 33 33 33 02 03 09 06 01 01 01 07 00 FF 06 00 00 01 CA 02 02 0F 00 16 1B 02 03 09 06 01 01 02 07 00 FF 06 00 00 00 00 02 02 0F 00 16 1B 02 03 09 06 01 01 01 08 00 FF 06 00 16 79 50 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 00 FF 06 00 14 B9 D9 02 02 0F 00 16 1E 02 03 09 06 01 00 20 07 00 14 3A 7E 7E A8 A4 CF 02 23 03 99 96 FF 12 09 67 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 00 FF 12 09 66 02 02 0F FF 16 23 02 03 09 06 01 00 48 07 00 FF 12 09 6F 02 02 0F FF 16 23 02 03 09 06 01 00 1F 07 00 FF 12 00 BB 02 02 0F FE 16 21 02 03 09 06 01 00 33 07 00 FF 12 00 10 02 02 0F FE 16 21 02 03 09 06 01 00 47 07 00 FF 12 00 3D 02 02 0F FE 16 21 02 03 09 06 01 01 01 08 01 FF 06 00 0A B5 0C 02 02 0F 00 16 1E 02 03 09 06 01 01 01 08 02 FF 06 00 0B C4 44 02 02 0F 00 16 1E 02 03 09 06 01 01 02 05 98 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:asyncio:poll took 3.784 ms: 1 events
DEBUG:smartmeter:HDLC Buffer: 00 7E 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 0F 00 1C 38 10 0C 07 E6 08 12 04 15 1B 32 00 FF 88 80 02 10 02 02 09 06 00 00 60 01 00 FF 09 08 37 38 34 34 37 30 30 38 02 02 09 06 00 00 60 01 01 FF 09 06 31 32 39 33 33 33 02 03 09 06 01 01 01 07 00 FF 06 00 00 01 CA 02 02 0F 00 16 1B 02 03 09 06 01 01 02 07 00 FF 06 00 00 00 00 02 02 0F 00 16 1B 02 03 09 06 01 01 01 08 00 FF 06 00 16 79 50 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 00 FF 06 00 14 B9 D9 02 02 0F 00 16 1E 02 03 09 06 01 00 20 07 00 14 3A 7E 7E A8 A4 CF 02 23 03 99 96 FF 12 09 67 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 00 FF 12 09 66 02 02 0F FF 16 23 02 03 09 06 01 00 48 07 00 FF 12 09 6F 02 02 0F FF 16 23 02 03 09 06 01 00 1F 07 00 FF 12 00 BB 02 02 0F FE 16 21 02 03 09 06 01 00 33 07 00 FF 12 00 10 02 02 0F FE 16 21 02 03 09 06 01 00 47 07 00 FF 12 00 3D 02 02 0F FE 16 21 02 03 09 06 01 01 01 08 01 FF 06 00 0A B5 0C 02 02 0F 00 16 1E 02 03 09 06 01 01 01 08 02 FF 06 00 0B C4 44 02 02 0F 00 16 1E 02 03 09 06 01 01 02 05 98 7E 7E A0 2D CF 02 23 13 71 7C 08 01 FF 06 00 0E 49 DE 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 02 FF 06 00 06 6F FB 02 02 0F 00 16 1E 8B 3E 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
INFO:root:App shutting down now.
DEBUG:asyncio:Close <_UnixSelectorEventLoop running=False closed=False debug=True>
Traceback (most recent call last):
  File "/usr/bin/smartmeter-datacollector", line 14, in <module>
    app.main()
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/app.py", line 69, in main
    asyncio.run(build_and_start(app_config), debug=debug_mode)
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/app.py", line 29, in build_and_start
    data_collector.process_queue())
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 46, in start
    await self._serial.start_and_listen()
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/smartmeter/serial_reader.py", line 45, in start_and_listen
    self._callback(data)
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 57, in _data_received
    dlms_objects = self._parser.parse_to_dlms_objects()
  File "/usr/lib/python3/dist-packages/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py", line 84, in parse_to_dlms_objects
    parsed_objects = self._client.parsePushObjects(self._dlms_data.value[0])
  File "/home/pi/.local/lib/python3.7/site-packages/gurux_dlms/GXDLMSClient.py", line 1342, in parsePushObjects
[... cut ...]
  File "/home/pi/.local/lib/python3.7/site-packages/gurux_dlms/GXDLMSConverter.py", line 97, in __updateOBISCodeInfo
    list_ = codes.find(ln, it.objectType)
  File "/home/pi/.local/lib/python3.7/site-packages/gurux_dlms/GXStandardObisCodeCollection.py", line 62, in find
    return self.find2(self.getBytes(ln), objectType)
  File "/home/pi/.local/lib/python3.7/site-packages/gurux_dlms/GXStandardObisCodeCollection.py", line 51, in getBytes
    raise ValueError("Invalid OBIS Code.")
ValueError: Invalid OBIS Code.
```

xens, 2022-08-18:
In doubt I've also updated the version of gurux-dlms from 1.0.107 to 1.0.129 but it doesn't help, still have the same error message. I've also tried to specify a decryption key, the same that was specified inside DLMS Director (BlockCipher key: 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F) but again result is the same except the warning on top

```
WARNING:smartmeter:Using the Iskra AM550 meter with encrypted data has NOT BEEN TESTED yet!
INFO:smartmeter:Successfully set up Iskra AM550 smart meter on '/dev/ttyUSB0'.
```

raymar9, 2022-08-19:
Please read my comment here: https://github.com/scs/smartmeter-datacollector/issues/34#issuecomment-1220833858
I think you have the same problem as MarkusTeufelberger.

xens, 2022-08-22:
@MarkusTeufelberger thanks for the code sample / hints inside #34, it helped a lot.

It looks like my setup is a bit different, obis code and data are sent part of the same frame but I'm not sure that there's a time-stamp at all.

With the following code I'm able to parse the obis code + extract the associated value.

```python3
def parse_to_dlms_objects(self) -> Dict[str, GXDLMSObject]:
    parsed_objects: List[Tuple[GXDLMSObject, int]] = []
    if isinstance(self._dlms_data.value, list):
        p =  GXDLMSPushSetup()
        p.pushObjectList.append((GXDLMSClock(), GXDLMSCaptureObject(2, 0)))
        for idx in range(1,len(self._dlms_data.value),1):
            assert isinstance(self._dlms_data.value[idx][0], bytearray)
            assert len(self._dlms_data.value[idx][0]) == 6
            obis_code = ".".join(map(str, list(self._dlms_data.value[idx][0])))
            p.pushObjectList.append((GXDLMSRegister(obis_code), GXDLMSCaptureObject(2, 0)))
            print("raw-data: ", self._dlms_data.value[idx])
            print("parsed-data: ", obis_code, self._dlms_data.value[idx][1])
        parsed_objects = p.pushObjectList
    self._dlms_data.clear()
    return {obj.getName(): obj for obj, _ in parsed_objects}
```

As you can see in the output, I was not able to set the associated value for each obis code, which leads to the following errors:

```
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
raw-data:  [bytearray(b'\x00\x00`\x01\x01\xff'), bytearray(b'129333')]
parsed-data:  0.0.96.1.1.255 bytearray(b'129333')
raw-data:  [bytearray(b'\x01\x01\x01\x07\x00\xff'), 2145, [0, 27]]
parsed-data:  1.1.1.7.0.255 2145
raw-data:  [bytearray(b'\x01\x01\x02\x07\x00\xff'), 0, [0, 27]]
parsed-data:  1.1.2.7.0.255 0
raw-data:  [bytearray(b'\x01\x01\x01\x08\x00\xff'), 1496116, [0, 30]]
parsed-data:  1.1.1.8.0.255 1496116
raw-data:  [bytearray(b'\x01\x01\x02\x08\x00\xff'), 1383663, [0, 30]]
parsed-data:  1.1.2.8.0.255 1383663
raw-data:  [bytearray(b'\x01\x00 \x07\x00\xff'), 2347, [-1, 35]]
parsed-data:  1.0.32.7.0.255 2347
raw-data:  [bytearray(b'\x01\x004\x07\x00\xff'), 2393, [-1, 35]]
parsed-data:  1.0.52.7.0.255 2393
raw-data:  [bytearray(b'\x01\x00H\x07\x00\xff'), 2387, [-1, 35]]
parsed-data:  1.0.72.7.0.255 2387
raw-data:  [bytearray(b'\x01\x00\x1f\x07\x00\xff'), 826, [-2, 33]]
parsed-data:  1.0.31.7.0.255 826
raw-data:  [bytearray(b'\x01\x003\x07\x00\xff'), 15, [-2, 33]]
parsed-data:  1.0.51.7.0.255 15
raw-data:  [bytearray(b'\x01\x00G\x07\x00\xff'), 97, [-2, 33]]
parsed-data:  1.0.71.7.0.255 97
raw-data:  [bytearray(b'\x01\x01\x01\x08\x01\xff'), 707812, [0, 30]]
parsed-data:  1.1.1.8.1.255 707812
raw-data:  [bytearray(b'\x01\x01\x01\x08\x02\xff'), 788304, [0, 30]]
parsed-data:  1.1.1.8.2.255 788304
raw-data:  [bytearray(b'\x01\x01\x02\x08\x01\xff'), 946324, [0, 30]]
parsed-data:  1.1.2.8.1.255 946324
raw-data:  [bytearray(b'\x01\x01\x02\x08\x02\xff'), 437339, [0, 30]]
parsed-data:  1.1.2.8.2.255 437339
DEBUG:smartmeter:Unable to find ID object. Using fallback ID /dev/ttyUSB0.
WARNING:smartmeter:Unable to parse timestamp from 0.0.1.0.0.255. Using system time.
WARNING:smartmeter:No value received for 1.1.1.7.0.255.
WARNING:smartmeter:No value received for 1.1.2.7.0.255.
WARNING:smartmeter:No value received for 1.1.1.8.0.255.
WARNING:smartmeter:No value received for 1.1.2.8.0.255.
WARNING:smartmeter:No value received for 1.0.32.7.0.255.
WARNING:smartmeter:No value received for 1.0.52.7.0.255.
WARNING:smartmeter:No value received for 1.0.72.7.0.255.
WARNING:smartmeter:No value received for 1.0.31.7.0.255.
WARNING:smartmeter:No value received for 1.0.51.7.0.255.
WARNING:smartmeter:No value received for 1.0.71.7.0.255.
WARNING:smartmeter:No value received for 1.1.1.8.1.255.
WARNING:smartmeter:No value received for 1.1.1.8.2.255.
WARNING:smartmeter:No value received for 1.1.2.8.1.255.
WARNING:smartmeter:No value received for 1.1.2.8.2.255.
```

I tried to play with `self._client.updateValue(obj, attr_ind, self._dlms_data.value[index])` without success, also I'm not sure if there's a way to define the obis code + the value directly in one go without doing it separately ex using something similar to `p.pushObjectList.append((GXDLMSRegister(obis_code), GXDLMSCaptureObject(2, 0)))` but passing also the value.

MarkusTeufelberger, 2022-08-22:
Could you post a decrypted data sample? Also you need to add the values to your objects, in your code you currently just create an object full of obis codes but never assign any values to it (the `self._client.updateValue()` part)

MarkusTeufelberger, 2022-08-23:
You can get this by adding a `print(self._parser._dlms_data)` after https://github.com/scs/smartmeter-datacollector/blob/d4613f504b7b0e1bb63af14c67f200402a6db744/smartmeter_datacollector/smartmeter/meter.py#L56 for example.

xens, 2022-08-23:
Here you go:

```
02 10 02 02 09 06 00 00 60 01 00 FF 09 08 37 38 34 34 37 30 30 38 02 02 09 06 00 00 60 01 01 FF 09 06 31 32 39 33 33 33 02 03 09 06 01 01 01 07 00 FF 06 00 00 00 00 02 02 0F 00 16 1B 02 03 09 06 01 01 02 07 00 FF 06 00 00 02 F5 02 02 0F 00 16 1B 02 03 09 06 01 01 01 08 00 FF 06 00 16 E3 ED 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 00 FF 06 00 15 20 5B 02 02 0F 00 16 1E 02 03 09 06 01 00 20 07 00 FF 12 09 9A 02 02 0F FF 16 23 02 03 09 06 01 00 34 07 00 FF 12 09 80 02 02 0F FF 16 23 02 03 09 06 01 00 48 07 00 FF 12 09 81 02 02 0F FF 16 23 02 03 09 06 01 00 1F 07 00 FF 12 01 9C 02 02 0F FE 16 21 02 03 09 06 01 00 33 07 00 FF 12 00 0B 02 02 0F FE 16 21 02 03 09 06 01 00 47 07 00 FF 12 00 5F 02 02 0F FE 16 21 02 03 09 06 01 01 01 08 01 FF 06 00 0A D2 F7 02 02 0F 00 16 1E 02 03 09 06 01 01 01 08 02 FF 06 00 0C 10 F6 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 01 FF 06 00 0E 74 00 02 02 0F 00 16 1E 02 03 09 06 01 01 02 08 02 FF 06 00 06 AC 5B 02 02 0F 00 16 1E
``` 

> Also you need to add the values to your objects, in your code you currently just create an object full of obis codes but never assign any values to it (the self._client.updateValue() part)

Yes I'm aware I removed that part temporarily from the code I've pasted above, I wasn't able to properly leverage `self._client.updateValue()` and I was wondering if it was possible to assign the value directly when adding the obis code instead of doing two different iterations (for the obis code and then for the values).

xens, 2026-09-06:
This meter still does not work on v2.0.0, and I now have the cause and a fix.

I tried to reopen this issue and GitHub would not let me, so **could a maintainer reopen it** (or tell me to file a fresh one and I will)?

Context for anyone arriving fresh — this issue was closed in May 2023 alongside #48 *Parse data
without push setup*. That added `_parse_dlms_without_push_list()`, which handles a **flat**
`[obis, value, obis, value, …]` message. This meter sends a **nested** one, so it was never
covered. Same meter as in 2022: serial `78447008`, the one in the capture I posted then.

### What happens on v2.0.0

```
File ".../smartmeter/hdlc_dlms_parser.py", line 104, in parse_to_dlms_objects
    dlms_objects = self._parse_dlms_with_push_object_list()
File ".../smartmeter/hdlc_dlms_parser.py", line 166, in _parse_dlms_with_push_object_list
    parsed_objects = self._client.parsePushObjects(self._dlms_data.value[0])
File ".../gurux_dlms/GXStandardObisCodeCollection.py", line 51, in getBytes
    raise ValueError("Invalid OBIS Code.")
```

Nothing catches it, so the reader task dies and the process exits — a restart loop under Docker.

### Cause

The meter sends no push-object-list and wraps every OBIS code and value in a structure of its own,
optionally followed by the register's `[scaler, unit]`:

```
value    = list[16]
value[0] = [bytes[6] 0-0:96.1.0*255, bytes[8] b'78447008']   <- meter ID, a data structure
value[2] = [bytes[6] 1-1:1.7.0*255, 0, [0, 27]]              <- value + [scaler, unit]
```

The parser selects its branch with `isinstance(self._dlms_data.value[0], list)`. Here `value[0]`
*is* a list, so the message is taken for one carrying a push-object-list and the OBIS
octet-string reaches `parsePushObjects()`, where a capture object is expected.

A genuine push-object-list is an array of capture-object structures
`[class_id, logical_name, attribute_index, data_index]`, so its own first element is itself a
structure — which separates the two layouts exactly:

| | `value[0]` | `value[0][0]` |
|---|---|---|
| your sample `UNENCRYPTED_VALID_DATA` | `list[28]` of capture objects | **`list[4]`** |
| this meter | `[obis, value]` | **`bytes[6]`** |

### Fix

PR: xens/smartmeter-datacollector `fix/am550-no-push-object-list` (opening against master).

- Detect a push-object-list by that shape, and teach `extract_obis_and_values()` the grouped
  layout. Meters that work today keep exactly the path they take today.
- The frame's `[scaler, unit]` is **ignored on purpose**: scaling stays in the COSEM register
  configuration as on the push-object-list path. Honouring it would scale twice — `VOLTAGE_L1`
  has a `-1` scaler in the frame *and* `RegisterCosem(…, 0.1)` in `iskraam550.py`.
- Also decodes octet-string meter IDs in `retrieve_id()`, which on this path produced
  `"bytearray(b'78447008')"` — that would be the meter ID in every MQTT topic.

Verification: your suite goes **53 → 54** with no existing test modified; reverting the source
change makes the new test fail with the exact `ValueError` above; and against 60 s of live frames
**12/12 messages decode** with correct SI values (`231.7 V`, `10.25 A`) and counters matching the
meter's display.

I have the meter and am happy to test any alternative you'd prefer — for instance catching the
`ValueError` and falling back. I chose explicit shape detection because a fallback would also
swallow genuinely malformed push-object-lists.

raymar9, 2026-10-20:
Hi xens, sorry for the late response. Some other projects and holidays kept me from looking into this issue and your proposed PR. I will gladly do that in the next couple of days. Thanks for diving into this problem and providing a solution.

closed 2026-11-02 by raymar9