scs/smartmeter-datacollector #35: No data for Iskraemeco AM550

mainzelM, 2022-08-31:
I setup the smartmeter-datacollector using the provided raspberry Pi image. I added both sinks, but there is no data in grafana or in the log:
```
root@raspberrypi:/var/log# journalctl -u DataLogger
-- Logs begin at Wed 2022-08-31 13:14:13 CEST, end at Wed 2022-08-31 14:00:04 CEST. --
-- No entries --
```

My smartmeter is an Iskraemeco AM550 operated by the EKZ.

How can I debug this?

raymar9, 2022-09-07:
Hi mainzelM
You don't see the log with `journalctl` because the unit-name `DataLogger` of the service is wrong. Use `journalctl -u python3-smartmeter-datacollector` instead.

It seems that no serial data arrive at the Raspberry Pi. This can have multiple reasons. I would start with a simple serial reader tool like minicom (or Putty) - either on the Raspberry Pi or attach the smart-meter to a computer and check if any serial data is incoming.

mainzelM, 2022-09-08:
Hi raymar9

Thanks for your reply! In the meantime the EKZ support team has contacted me: my smartmeter needs a firmware update before I can access it. The update should arrive within the next two months.

closed 2022-09-08 by mainzelM