scs/smartmeter-datacollector #39: Cannot setup dev environment with python 3.10 on ubuntu 22.04

kreutpet, 2022-09-19:
i am trying to setup the development environment but i cannot get the command pipenv command running

```
pipenv install --dev
Traceback (most recent call last):
  File "/usr/bin/pipenv", line 33, in <module>
    sys.exit(load_entry_point('pipenv==11.9.0', 'console_scripts', 'pipenv')())
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 696, in main
    with self.make_context(prog_name, args, **extra) as ctx:
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 621, in make_context
    self.parse_args(ctx, args)
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 1018, in parse_args
    rest = Command.parse_args(self, ctx, args)
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 875, in parse_args
    parser = self.make_parser(ctx)
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 821, in make_parser
    for param in self.get_params(ctx):
  File "/usr/lib/python3/dist-packages/pipenv/vendor/click/core.py", line 774, in get_params
    help_option = self.get_help_option(ctx)
  File "/usr/lib/python3/dist-packages/pipenv/cli.py", line 26, in get_help_option
    from .import core
  File "/usr/lib/python3/dist-packages/pipenv/core.py", line 21, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/pipenv/vendor/requests/__init__.py", line 65, in <module>
    from . import utils
  File "/usr/lib/python3/dist-packages/pipenv/vendor/requests/utils.py", line 27, in <module>
    from .cookies import RequestsCookieJar, cookiejar_from_dict
  File "/usr/lib/python3/dist-packages/pipenv/vendor/requests/cookies.py", line 172, in <module>
    class RequestsCookieJar(cookielib.CookieJar, collections.MutableMapping):
AttributeError: module 'collections' has no attribute 'MutableMapping'

```
what is the best way to handle this ?

kreutpet, 2022-09-19:
during the install i do get an error message 
```
sudo apt install ./python3-smartmeter-datacollector_1.0.2-1_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-smartmeter-datacollector' instead of './python3-smartmeter-datacollector_1.0.2-1_amd64.deb'
The following NEW packages will be installed:
  python3-smartmeter-datacollector
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/17.6 kB of archives.
After this operation, 105 kB of additional disk space will be used.
Get:1 /root/python3-smartmeter-datacollector_1.0.2-1_amd64.deb python3-smartmeter-datacollector all 1.0.2-1 [17.6 kB]
Selecting previously unselected package python3-smartmeter-datacollector.
(Reading database ... 30034 files and directories currently installed.)
Preparing to unpack .../python3-smartmeter-datacollector_1.0.2-1_amd64.deb ...
Unpacking python3-smartmeter-datacollector (1.0.2-1) ...
Setting up python3-smartmeter-datacollector (1.0.2-1) ...
Installing dependencies using pip....done
N: Download is performed unsandboxed as root as file '/root/python3-smartmeter-datacollector_1.0.2-1_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
```
could that also be an issue ?

Montellese, 2022-09-20:
Please see https://github.com/pypa/pipenv/issues/5088 and give it a try. It looks like the version of `pipenv` provided by Ubuntu packages is outdated and not compatible with Python 3.10. Try uninstalling the APT package and use `pip install pipenv` to install `pipenv`.

kreutpet, 2022-09-20:
many thanks .
i tied to install the pipenv with recommend command but end up in another issue.
python 3.10 is installed but when running the command 3.8 is needed.
```
root@dlms:~/smartmeter-datacollector# /usr/local/bin/pipenv install --dev
Warning: Python 3.8 was not found on your system...
Neither 'pyenv' nor 'asdf' could be found to install Python.
You can specify specific versions of Python with:
$ pipenv --python path/to/python
```
I could not find any specific dependency in your code.

raymar9, 2022-09-21:
But there is a requirement in the Pipfile:
```
[requires]
python_version = "3.8"
```
You can try using another Python version by initializing the environment with:
```
pipenv --python three install --dev
OR
pipenv --python 3.10 install --dev
OR
pipenv --python <path to Python> install --dev
```
This should override the setting in the Pipfile. Please see https://pipenv-fork.readthedocs.io/en/latest/basics.html#specifying-versions-of-python for further information.

You can also comment out the line in the Pipfile for your setup if the above methods do not work.

We specified version 3.8 because the Raspbian distribution we use for the demo has Python 3.8 installed by default.

closed 2022-09-21 by raymar9

kreutpet, 2022-09-21:
actually i gave up on ubuntu 22.04 and installed 20.4 which has python3.8
i guess sooner or later t will be solved for 22.04
i do have it up and running now in 20.04. 
it is the first time i use pipenv. What would be the next steps to start the smartmeter-datacollector manually for debugging?
i assume pipenv shell would be need to enter into the environment?

thanks for the support

raymar9, 2022-09-22:
`pipenv shell` sets up the shell with the correct Python environment (virtualenv) and its dependencies which you previously installed by `pipenv install`. 
Then you start the software with `python -m smartmeter-datacollector -c <config.ini path>`. Please see the [README ](https://github.com/scs/smartmeter-datacollector/blob/master/README.md) or our [Wiki](https://github.com/scs/smartmeter-datacollector/wiki/How-to-use) for more information.

kreutpet, 2022-09-24:
many thanks. i missed the pipenv install.