scs/smartmeter-datacollector #40: Wrong unit of current value

sparx82, 2022-09-28:
I'm using smartmeter-datacollector to collect data from an Landis + Gyr E450 provided by AEW. For the current values (L1 - L3), I get the following data:

`DEBUG:smartmeter:3 1.0.31.7.0.255 2: 106`

which transforms into:

`INFO:DataLogger:58161286 - 2022-09-28T06:17:09.945004 - Current L1: 106.0 A`

This is definitely wrong as this should be 1.06A. Same for the other current values. All the other values are ok. Is this a electricity provider specific thing or is this wrong in smartmeter-datacollector?

raymar9, 2022-09-29:
This is interesting. I would have assumed that the meter sends the values scaled by the SI unit. Also the OBIS code standard for *.31.7.* says that the values are sent in amperes. In your case it would be in 0.01A. 
However, we haven't tested the software with a L+G E450 meter that provides the current L1,2,3, so it is possible that every L+G E450 meter uses this scaling. We will keep that in mind when we get another L+G E450 meter for testing. Thanks for this contribution.
You can however fix this issue on your device where you installed the smartmeter-datacollector. Open the file `/usr/lib/python3/dist-packages/smartmeter_datacollector/smartmeter/lge450.py` and edit the lines 35, 43 and 51 e.g.

`RegisterCosem("1.0.31.7.0.255", MeterDataPointTypes.CURRENT_L1.value),` to
`RegisterCosem("1.0.31.7.0.255", MeterDataPointTypes.CURRENT_L1.value, 0.01),`

This overrides the default scaling factor of `1.0`. Save the file and restart the smartmeter-datacollector with `systemctl restart python3-smartmeter-datacollector`.

sparx82, 2022-09-30:
Thanks for providing this solution. If I can be of any other help (e.g. providing data), just tell me.

What I saw is that the data in this issue also contains current in 0.01A
https://github.com/scs/smartmeter-datacollector/issues/31#issuecomment-1251370795

IMHO this makes more sense because if it would be in Amperes, the granularity would be too coarse.

raymar9, 2023-05-15:
Bug will be fixed in next Release v1.1.0 .

closed 2023-05-15 by raymar9