scs/smartmeter-datacollector #45: Landis+Gyr E450 Energienetzte Steiermark - Failed to extract data from HDLC frame: 'Secure connection is not supported.'

cweigl, 2023-01-16:
When i am starting "smartmeter-datacollector" i receive always the message below. Is something wrong with the key ?
(GUEK - 32 Zeichen - 16Byte = 128Bit "3AB2DFD6A58981FE626AA6DEC4D566EE") - i added the key direct in config.py
The HDLC Buffer has not always the same length,.? May be someone has an idea - i used the fork from Martink173
Thank you for any help.

`$ smartmeter-datacollector
WARNING:root:No config file found. Using default config.
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/ttyUSB0'.
ERROR:sink:Operation timed out
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 01 FF C4 FD DB 1E BE F7 F1 61 30 D0 BD 59 A2 90 C1 D7 2A 29 66 D2 03 0F 09 74 0D 16 81 23 22 20 47 34 CB 00 62 E8 61 7A 85 60 90 ED 84 8A CD BE D1 1A A9 97 E2 48 92 3D 78 4A 3C 26 0F 97 FF B4 03 38 06 66 9E FC 21 DE CC D4 48 74 AF FD 97 9D 5F 49 11 D0 8D 1B C7 0C 21 CE A3 11 C0 92 51 E7 95 A9 B7 66 90 06 96 64 FE 21 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 01 FF C4 FD DB 1E BE F7 F1 61 30 D0 BD 59 A2 90 C1 D7 2A 29 66 D2 03 0F 09 74 0D 16 81 23 22 20 47 34 CB 00 62 E8 61 7A 85 60 90 ED 84 8A CD BE D1 1A A9 97 E2 48 92 3D 78 4A 3C 26 0F 97 FF B4 03 38 06 66 9E FC 21 DE CC D4 48 74 AF FD 97 9D 5F 49 11 D0 8D 1B C7 0C 21 CE A3 11 C0 92 51 E7 95 A9 B7 66 90 06 96 64 FE 21 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 2D DB EA E3 3A 4B DD AD F7 3D 36 FD 26 24 A4 F5 E0 B3 A8 B5 77 15 0B AC AF 1A C3 18 8C 9E 43 29 F9 83 84 4E BE 7F FB DF FD 20 59 A0 EA 7A E9 76 B2 41 D8 9F 63 CF DA 13 FB 95 46 C8 46 9C 98 0A CF B5 ED 58 86 AF 36 1F 32 0E 0A 7D 7E
DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 01 FF C4 FD DB 1E BE F7 F1 61 30 D0 BD 59 A2 90 C1 D7 2A 29 66 D2 03 0F 09 74 0D 16 81 23 22 20 47 34 CB 00 62 E8 61 7A 85 60 90 ED 84 8A CD BE D1 1A A9 97 E2 48 92 3D 78 4A 3C 26 0F 97 FF B4 03 38 06 66 9E FC 21 DE CC D4 48 74 AF FD 97 9D 5F 49 11 D0 8D 1B C7 0C 21 CE A3 11 C0 92 51 E7 95 A9 B7 66 90 06 96 64 FE 21 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 2D DB EA E3 3A 4B DD AD F7 3D 36 FD 26 24 A4 F5 E0 B3 A8 B5 77 15 0B AC AF 1A C3 18 8C 9E 43 29 F9 83 84 4E BE 7F FB DF FD 20 59 A0 EA 7A E9 76 B2 41 D8 9F 63 CF DA 13 FB 95 46 C8 46 9C 98 0A CF B5 ED 58 86 AF 36 1F 32 0E 0A 7D 7E 02 2A E5 4D 9D 8A 3E 8D 05 F6 FD B8 AC AB 9D 95 99 C2 25 B0 A4 C2 E5 51 E3 E8 29 73 96 34 27 71 B1 D1 57 89 90 A8 E1 2F 30 F5 74 3F 96 B0 25 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 01 FF C4 FD DB 1E BE F7 F1 61 30 D0 BD 59 A2 90 C1 D7 2A 29 66 D2 03 0F 09 74 0D 16 81 23 22 20 47 34 CB 00 62 E8 61 7A 85 60 90 ED 84 8A CD BE D1 1A A9 97 E2 48 92 3D 78 4A 3C 26 0F 97 FF B4 03 38 06 66 9E FC 21 DE CC D4 48 74 AF FD 97 9D 5F 49 11 D0 8D 1B C7 0C 21 CE A3 11 C0 92 51 E7 95 A9 B7 66 90 06 96 64 FE 21 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 2D DB EA E3 3A 4B DD AD F7 3D 36 FD 26 24 A4 F5 E0 B3 A8 B5 77 15 0B AC AF 1A C3 18 8C 9E 43 29 F9 83 84 4E BE 7F FB DF FD 20 59 A0 EA 7A E9 76 B2 41 D8 9F 63 CF DA 13 FB 95 46 C8 46 9C 98 0A CF B5 ED 58 86 AF 36 1F 32 0E 0A 7D 7E 02 2A E5 4D 9D 8A 3E 8D 05 F6 FD B8 AC AB 9D 95 99 C2 25 B0 A4 C2 E5 51 E3 E8 29 73 96 34 27 71 B1 D1 57 89 90 A8 E1 2F 30 F5 74 3F 96 B0 25 7E 7E A0 30 CE FF 03 13 86 F8 E0 C0 00 03 00 00 1F 12 5E FA F4 20 28 E5 94 B5 1B AD E2 A1 09 A6 11 20 B5 8E 75 DF 53 47 33 27 5E BE B1 47 61 4A 1D EF 7E
WARNING:smartmeter:Failed to extract data from HDLC frame: 'Secure connection is not supported.' Some data got lost.`

raymar9, 2023-01-17:
Your key seems to be working. I tested it with the [Gurux DLMS Translator](http://www.gurux.fi/gurux_dlms_translator45/Gurux_dlms_translator.html).
But somehow the key is not handed over (correctly) to the HDLC Parser and thus, the warning `Secure connection is not supported.` is shown. In other words, an unencrypted parser (without your key) is used instead of the encrypted which you need. It does not matter that the HDLC buffer has not always the same length.

Why do you use the `config.py` (default config) and not the `config.ini` file? The `.ini` File should look like:
```
[reader0]
type = lge450
port = /dev/ttyUSB0
baudrate = 2400
key = 3AB2DFD6A58981FE626AA6DEC4D566EE
```

cweigl, 2023-01-18:
I think, you mean the datacollector.ini file. If this file exists, i get an error: 
` smartmeter-datacollector
Traceback (most recent call last):
  File "/usr/local/bin/smartmeter-datacollector", line 14, in <module>
    app.main()
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/app.py", line 69, in main
    asyncio.run(build_and_start(app_config), debug=debug_mode)
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/app.py", line 27, in build_and_start
    await asyncio.gather(
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 46, in start
    await self._serial.start_and_listen()
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/smartmeter/serial_reader.py", line 45, in start_and_listen
    self._callback(data)
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/smartmeter/meter.py", line 57, in _data_received
    dlms_objects = self._parser.parse_to_dlms_objects()
  File "/usr/local/lib/python3.10/dist-packages/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py", line 84, in parse_to_dlms_objects
    parsed_objects = self._client.parsePushObjects(self._dlms_data.value[0])
  File "/usr/local/lib/python3.10/dist-packages/gurux_dlms/GXDLMSClient.py", line 1257, in parsePushObjects
    classID = tmp[0]
TypeError: 'int' object is not subscriptable`

And when i delete datacollector.ini file, i get: 

`smartmeter-datacollector
WARNING:root:No config file found. Using default config.
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/ttyUSB0'.
ERROR:sink:Operation timed out
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 02 38 0F 47 AC 45 67 7C 02 9D 28 45 40 66 58 7F 77 73 8A FB 40 13 06 8D D0 40 43 54 6B 05 0C 48 88 8D 73 D9 90 90 E8 93 A1 AB F1 AE 22 D2 C5 FC 7D 4E 32 47 60 ED 13 B4 96 78 4B DF 8D A9 DC 93 F8 55 4A C6 73 13 9F 43 80 99 F8 F5 25 58 74 1B 9D FA 11 AE C6 10 17 BC B5 7C 82 E9 87 AE E2 52 61 3E 34 4C A3 6A 60 E0 89 E8 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 02 38 0F 47 AC 45 67 7C 02 9D 28 45 40 66 58 7F 77 73 8A FB 40 13 06 8D D0 40 43 54 6B 05 0C 48 88 8D 73 D9 90 90 E8 93 A1 AB F1 AE 22 D2 C5 FC 7D 4E 32 47 60 ED 13 B4 96 78 4B DF 8D A9 DC 93 F8 55 4A C6 73 13 9F 43 80 99 F8 F5 25 58 74 1B 9D FA 11 AE C6 10 17 BC B5 7C 82 E9 87 AE E2 52 61 3E 34 4C A3 6A 60 E0 89 E8 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 64 2B E1 57 48 69 F6 39 8F 1C 37 6F 2E FC 99 FB DA 35 99 3A 94 72 5B A7 99 F5 BD 79 5D 58 DB 34 B2 46 B5 81 2C 7B 2C 66 86 FA 5A 33 4B 74 7A 69 73 91 71 7C 89 0F 03 51 D2 DA 93 D6 5B E8 A9 32 2D EA 19 60 2A 62 D0 77 DF 2E 29 0C 87 F1 68 D0 4F 28 F6 FF 15 44 08 AC DA 13 27 5F F4 A3 40 78 59 C0 35 FB E3 2F A6 48 01 15 DA 19 D3 62 9E B5 CD 3A 79 79 5D 2E E1 C8 D7 BB 55 2D 7E
DEBUG:smartmeter:More DLMS data expected. Not yet ready to be parsed.
DEBUG:smartmeter:HDLC Buffer: 7E A0 8B CE FF 03 13 EE E1 E6 E7 00 E0 40 00 01 00 00 77 DB 08 4C 47 5A 67 73 7C 51 BD 82 01 03 30 00 02 38 0F 47 AC 45 67 7C 02 9D 28 45 40 66 58 7F 77 73 8A FB 40 13 06 8D D0 40 43 54 6B 05 0C 48 88 8D 73 D9 90 90 E8 93 A1 AB F1 AE 22 D2 C5 FC 7D 4E 32 47 60 ED 13 B4 96 78 4B DF 8D A9 DC 93 F8 55 4A C6 73 13 9F 43 80 99 F8 F5 25 58 74 1B 9D FA 11 AE C6 10 17 BC B5 7C 82 E9 87 AE E2 52 61 3E 34 4C A3 6A 60 E0 89 E8 7E 7E A0 8B CE FF 03 13 EE E1 E0 40 00 02 00 00 7A 64 2B E1 57 48 69 F6 39 8F 1C 37 6F 2E FC 99 FB DA 35 99 3A 94 72 5B A7 99 F5 BD 79 5D 58 DB 34 B2 46 B5 81 2C 7B 2C 66 86 FA 5A 33 4B 74 7A 69 73 91 71 7C 89 0F 03 51 D2 DA 93 D6 5B E8 A9 32 2D EA 19 60 2A 62 D0 77 DF 2E 29 0C 87 F1 68 D0 4F 28 F6 FF 15 44 08 AC DA 13 27 5F F4 A3 40 78 59 C0 35 FB E3 2F A6 48 01 15 DA 19 D3 62 9E B5 CD 3A 79 79 5D 2E E1 C8 D7 BB 55 2D 7E 7E A0 30 CE FF 03 13 86 F8 E0 C0 00 03 00 00 1F 31 45 A7 9E 84 27 D9 41 E1 C3 07 FD 88 70 42 16 BC 69 04 BE 26 BC D3 17 16 E3 97 FE 48 5C AC C1 5D 7E
WARNING:smartmeter:Failed to extract data from HDLC frame: 'Secure connection is not supported.' Some data got lost.`

I checked again the raw Data, i received with the tool "Minicom" in ttyUSB0 - i get always 332Bytes like this :

`7e a0 8b ce ff 03 13 ee e1 e6 e7 00 e0 40 00 01 00 00 77 db 08 4c 47 5a 67 73 7c 51 bd 82 01 03 30 00 01 fe bf 47 bd f0 7e de 50 3c a7 9f 4a 27 8a 93 23 fe 6d 51 fc 53 7f 30 66 55 03 b4 2a 0d fb 6f 48 2c 0b 28 32 44 b0 7f 78 1d ad 51 de 21 de f6 fb 02 1a a1 23 62 ed 9c d1 60 71 03 ae a0 51 cf 74 99 15 c7 db e0 ca 54 93 3c 05 7a ae 49 0b 33 ee 1d 71 dc c2 3e a5 c7 01 f8 5f b7 ac 4d bd 04 63 24 8a b9 9f 74 f2 7e 0f c0 7e 7e a0 8b ce ff 03 13 ee e1 e0 40 00 02 00 00 7a 2d 3e 52 73 62 ce 92 61 d1 3e 0b e8 43 85 5a 29 9f c7 8d 25 64 06 d4 87 a5 2c 1d 98 b5 8b e5 c5 90 3c 4c f9 cd 16 d9 b7 be 93 2b 41 ff 86 d3 93 ad ef 30 2f c7 46 0d 44 7c 4f 57 8c 4b e2 88 4a b7 0b 74 9d 5d ef 11 f7 16 58 9b 66 d1 d3 9c 95 5e 3d ae 92 d7 78 f0 92 6c 35 da b0 b8 30 c7 c3 82 2d 46 1d 11 e2 e9 fd 16 3b 60 04 d9 bc 86 a9 d7 81 04 d9 c0 6f 5f 86 f6 db cd 7e 7e 7e a0 30 c [... cut ...]  26 6c 32 58 f7 c5 8c b5 42 fa 48 04 6d c0 a6 cc 91 7a 48 14 9e c0 61 f4 dd 0e d0 a7 44 7e 7b 25 95 da 62 92 cf df cf bb 7e 7b eb fe d0 3a 36 76 20 fc f1 70 91 50 a2 3a 0a 63 cd f1 69 81 7e 7e a0 30 ce ff 03 13 86 f8 e0 c0 00 03 00 00 1f f3 3b b5 3a cf fd 6d 0f ce a2 58 2f 81 53 a6 b6 a0 de d6 3c bd 61 13 5b ae 3b 48 48 97 26 71 50 ce 7e`

i was not able to iTranslate it, but a i have a Second key GAK: `BC37F8C87AEB07DA23393F4CAB3D70DA`
i do not know is the GAK needed as well or only the GUEK,.?

cweigl, 2023-01-19:
I even was not able to translate the raw Data with Gurux DLMS Translator, but i think the raw data are ok, here are the same raw datas:
#21 [https://github.com/scs/smartmeter-datacollector/issues/21](url)

How can i use the Raw data in the translator? May someone provides an exactly description with my raw datas,..
Then i have to fix:

`WARNING:smartmeter:Failed to extract data from HDLC frame: 'Secure connection is not supported.' Some data got lost.`

Thank you

closed 2023-08-28 by raymar9