scs/smartmeter-datacollector #54: L+G E450 Netz Burgenland unknown DLMS structure

rooneyaut, 2023-06-19:
I've installed the latest smartmeter-datacollector v1.1.0 on my Raspberry Pi and connected an MBUS-USB-Adapter to my E450.
The data from Netz Burgenland is encrypted with DLMS/COSEM regarding to their mail.
These are the settings specified from Netz Burgenland: 
```
Baud: 2400
Parity:  Even
```

Here is my datacollector.ini
```
[reader0]
type = lge450
port = /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BEA:n114J20-if00-port0
key = 6E3272357538782F443F442B472D4B64

[sink1]
host = 192.168.178.70
port = 1883
tls = False
check_hostname = False
password = mqtt.password
username = mqtt.user
type = mqtt

[sink0]
name = DataLogger
type = logger

[logging]
default = DEBUG
```

Here is the debug output:
```
DEBUG:asyncio:Using selector: EpollSelector
INFO:smartmeter:Successfully set up L+G E450 smart meter on '/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BEA:n114J20-if00-port0'.
INFO:sink:Connected to MQTT broker.
DEBUG:smartmeter:HDLC Buffer: 7E A0 6D CE FF 03 13 90 F1 E6 E7 00 DB 08 4C 47 5A 67 72 2C 8B D4 55 30 00 52 E7 62 2B 8D 52 7D 93 1F 18 E5 C4 AF 3D 1A 1C E5 75 7C A1 8F 22 50 C6 ED C3 57 64 D2 F5 2F 41 1A C2 B2 49 8A 37 46 B1 98 39 C1 CA 43 E9 E2 93 FC 07 F0 13 47 9C 27 F6 0A F8 62 FF 11 FD 8B C8 A9 D1 06 6A 8F F0 E0 E2 E9 BF B3 5C E9 6C 63 01 D6 1C 52 38 EC 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
WARNING:smartmeter:DLMS message is formatted in unknown structure and cannot be parsed by this software.
DEBUG:smartmeter:Unable to find ID object. Using fallback ID /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BEA:n114J20-if00-port0.
DEBUG:smartmeter:HDLC Buffer: 7E A0 6D CE FF 03 13 90 F1 E6 E7 00 DB 08 4C 47 5A 67 72 2C 8B D4 55 30 00 52 E7 63 66 BD 2D A5 79 7A 5F 31 7F 44 C3 0E 11 B9 57 74 4D ED B8 50 F4 2E 77 4F 5C BD 06 D2 9E 88 40 FF 06 EE FC D5 BA 95 C9 14 A2 A1 C4 36 CA 75 32 E4 57 7B BD 3B E9 7B A1 1C D5 8B 12 C3 A2 B3 E5 F0 D6 53 B7 EB 44 77 98 CC 9C 12 A1 72 E5 DD 99 E2 F3 37 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
WARNING:smartmeter:DLMS message is formatted in unknown structure and cannot be parsed by this software.
DEBUG:smartmeter:Unable to find ID object. Using fallback ID /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BEA:n114J20-if00-port0.
DEBUG:smartmeter:HDLC Buffer: 7E A0 6D CE FF 03 13 90 F1 E6 E7 00 DB 08 4C 47 5A 67 72 2C 8B D4 55 30 00 52 E7 64 BA 88 0A 40 78 00 56 72 5E 3E BF 97 3A 00 EB EE 8B 6B 8B E2 80 E0 CA 15 40 64 7C 54 8C 1F E3 B5 13 1A 3C FA 20 36 FC 1B CE B6 6B 4F B4 C8 A7 EC 3B 27 BD 84 0C 4B 39 EA E3 80 BA 16 50 E9 50 7B C5 5F 9D 09 46 07 B4 22 3C 56 EF 9B B9 C1 76 BD FD 34 7E
DEBUG:smartmeter:DLMS packet complete and ready for parsing.
WARNING:smartmeter:DLMS message is formatted in unknown structure and cannot be parsed by this software.
```

I hope that helps, I am trying to get the data from my E450 for several months now and then found this software and it looked really nice, but it's not working for some reason.

raymar9, 2023-06-20:
Hi @rooneyaut 
I tested your data samples both with the DLMS-Translator from Gurux (part of this software: https://www.gurux.fi/GXDLMSDirector) and with a test in our software. However I've had no success parsing the data.

DLMS Translator provides the following output:

```
<HDLC len="6C" >
<!-- Logical address:103, Physical address:127 -->
<TargetAddress Value="33FF" />
<SourceAddress Value="1" />
<!-- Notification frame. -->
<FrameType Value="13" />
<PDU>
<!-- IDIS system title:
Manufacturer Code: LGZ
Device type: IDIS package2 IP polyphase meter
Function type: Disconnector, Load Management, Multi Utility
Serial number: 36473812
 -->
  <GeneralGloCiphering>
    <SystemTitle Value="4C475A67722C8BD4" />
    <CipheredService Value="300052E7622B8D527D931F18E5C4AF3D1A1CE5757CA18F2250C6EDC35764D2F52F411AC2B2498A3746B19839C1CA43E9E293FC07F013479C27F60AF862FF11FD8BC8A9D1066A8FF0E0E2E9BFB35CE96C6301D61C52" />
  </GeneralGloCiphering>
</PDU>
</HDLC>
```

The HDLC layer seems to be valid, so it is no baudrate or serial transmission issue. But it can not decipher the data somehow. I do not why...
Maybe the key is wrong. Or you could ask in the Gurux forum https://www.gurux.fi/forum providing a data sample and your key.

raymar9, 2023-07-03:
Have you made any progress so far?
If you modified your key in the message above, it would explain a lot why I couldn't parse it.

I suspect however that your data does not contain any OBIS codes, only values and you need to setup the OBIS push setup yourself. See https://github.com/scs/smartmeter-datacollector/issues/34#issuecomment-1221298300

closed 2023-07-03 by raymar9