scs/smartmeter-datacollector #59: Smartmeter-datacollector failed to extract data from HDLC frame

urskle, 2024-01-28:
Smartmeter-datacollector shows warnings (smartmeter:Failed to extract data from HDLC frame: 'Invalid Command.' Some data got lost.) , then after a few minutes or hours (irregular intervals) an error (smartmeter:Failed to extract data from HDLC frame: 'Invalid Command.' Some data got lost.).
I have to restart the smartmeter-datacollector service, then the data are delivered without any problem for a few minutes or hours. Attached the log file.
I wonder why this behaviour is happening. I am running smartmeter-datacollector on a Raspberry Pi 3B with an ISKRA AM 550.
Any hints and suggestions are welcome!

[main.log](https://github.com/scs/smartmeter-datacollector/files/14076426/main.log)

raymar9, 2024-01-29:
Smartmeter-datacollector should recover after such warning or error and continue parsing data. To get more information on this matter I require some DEBUG log. Please modify the configuration (either manually in `/var/lib/smartmeter-datacollector/datacollector.ini` or with the web configurator) and set the log level to DEBUG (see also https://github.com/scs/smartmeter-datacollector/wiki/Configuration#logging-settings) Restart the software and wait until the problem appears again.

What version of the software do you use?

urskle, 2024-01-30:
Thank you for your fast reply!
Please find annexed a Screenshot of the grafana output. There are no values
displayed in Grafana after 21:40,. At the same time there is an error in
the logfile annexed. But according to the logfile the values are collected
even after 21:40 but are not entering influxdb. So the values cannot be
displayed in Grafana. I am very grateful for hints how to avoid that kind
of error.
I am using Smartmeter-datacollector v1.1.0 on a Raspberry
Pi3/Linux11(bullseye).

Jan 30 21:39:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jan 30 21:39:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 DB 08 49 53 4B 67 74 91 16 91 82 02 B5 20 00 0C 6D 29 29 FD 2D F3 37 17 8F F0 B1 15 18 0A 78 CA A8 47 2B 9D 86 E2 AD F6 E9 F3 EC 64 FF 36 9C CE A3 8A FB 7B D9 A9 CB 8B 48 49 9B EF 0B 33 EC 32 E0 EE F0 BE 72 D6 20 72 03 1A C3 07 FF 32 A1 8F 9A 61 09 E3 19 C5 80 72 75 D2 8B 1E F3 BF C1 78 48 6D 77 BC EE 61 4A A1 FC FA 16 D1 70 0D 83 28 94 1A 96 3C 39 04 A3 37 79 69 D3 ED D1 D7 02 86 4C 58 9A 67 24 B4 FC 18 84 37 7B 20 FD 68 B2 39 61 A2 D2 72 BD CE 77 DE 7C 7E 7E A8 A4 CF 02 23 03 99 96 B4 2C 81 20 F1 11 58 35 64 E0 B1 10 40 DB 28 37 4C BE 7C 6A 0D CC 6D F5 D8 FB DD 4E 5C 7E E5 83 14 4E 72 2D 16 4E 0E 23 6E 4C 52 73 84 25 72 BB 62 D7 CE 23 9E 7F E2 DC A4 A1 C5 80 89 77 A9 61 6C DC A5 D5 C2 99 C0 45 67 DD 4B 39 3F 16 72 FF 76 C0 55 94 F1 B7 E5 73 74 44 B2 57 AE B9 B2 96 99 E4 19 F0 5B A3 CC 30 08 B9 08 5A 7F 97 18 65 25 8D CE 3D 64 2A 81 C0 AA 37 84 B0 F6 00 C2 F2 6E 86 DE B5 B1 73 8C 7F 93 28 4C 0F F8 CD 20 8B E1 45 B7 9A F5 AF B0 4A 6B B3 AD 13 7E 7E A8 A4 CF 02 23 03 99 96 9B 70 75 63 24 D4 BE 23 E7 D7 8A 65 B4 9F D5 68 0A DD F3 20 94 19 69 E4 CB FC 08 CC A4 E5 F9 38 3F 27 4B 8A 69 7A 39 4D F1 08 FE 08 3B F1 73 58 24 49 76 9A EA 20 F6 EB 98 49 03 E4 FE 62 AE 76 70 FB 89 F2 80 B3 81 B7 49 92 46 86 9C B9 4E B9 7A 75 54 08 4C 4D AE DD D1 18 25 5D BF 4D 00 D9 94 70 8A 90 9E 82 C0 C4 D9 DA 21 15 85 AB C3 6A CF 6B DF A5 FC A5 E5 B1 75 AB 8E EA 19 AB D5 CC 68 38 CA FC 1D F9 E7 8E 88 F5 2F E4 57 89 B7 97 78 83 74 B6 4B 9C 79 46 BD D6 90 9E 7E 7E A8 A4 CF 02 23 03 99 96 0D 68 56 0D BD 30 F0 27 97 B1 F2 5A FB 37 E5 00 8D 15 EA 42 0C E0 F5 43 23 20 09 FF 56 88 07 A2 C1 F1 D9 16 87 80 28 66 AF 45 B1 CB BE 68 CF 26 B8 F0 B2 07 89 48 49 0C 81 1A E7 29 39 31 B1 21 59 9F 00 F8 21 15 FD 04 B1 03 87 9B C0 B3 A6 36 7E 46 92 AB 8C F5 E9 27 23 05 14 E5 96 B8 A4 4B A0 A1 A8 19 EF 3B CD B6 03 03 5D 70 DE D8 CE 3E FF B8 AF D3 EA 88 59 78 14 27 AE E4 7F 3A 4C 53 87 F0 59 2F 45 7C 1E 89 E8 A2 6C AE 84 0F 90 0B 0E 05 C1 A2 8D C1 13 6E 99 4E BD BE 7E 7E A0 67 CF 02 23 13 FB F1 F3 EE 35 74 A6 1D 8C 76 17 73 12 F3 A0 75 A8 B9 99 66 58 CA 43 BC 22 B0 72 7A A4 4E 2F 95 2A 58 BC 50 94 2E 6D 2E 91 5D F5 4B B4 6C 56 19 9C 7B 19 80 1C 13 1D D8 C0 2F 21 07 42 4E 88 75 70 5D 95 91 C0 76 C9 11 04 94 C6 9C B6 79 C4 3F 9D 2D B5 AF B5 81 E4 A8 85 EB EE 58 F6 56 55 68 7E
Jan 30 21:39:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 DB 08 49 53 4B 67 74 91 16 91 82 02 B5 20 00 0C 6D 29 29 FD 2D F3 37 17 8F F0 B1 15 18 0A 78 CA A8 47 2B 9D 86 E2 AD F6 E9 F3 EC 64 FF 36 9C CE A3 8A FB 7B D9 A9 CB 8B 48 49 9B EF 0B 33 EC 32 E0 EE F0 BE 72 D6 20 72 03 1A C3 07 FF 32 A1 8F 9A 61 09 E3 19 C5 80 72 75 D2 8B 1E F3 BF C1 78 48 6D 77 BC EE 61 4A A1 FC FA 16 D1 70 0D 83 28 94 1A 96 3C 39 04 A3 37 79 69 D3 ED D1 D7 02 86 4C 58 9A 67 24 B4 FC 18 84 37 7B 20 FD 68 B2 39 61 A2 D2 72 BD CE 77 DE 7C 7E 7E A8 A4 CF 02 23 03 99 96 B4 2C 81 20 F1 11 58 35 64 E0 B1 10 40 DB 28 37 4C BE 7C 6A 0D CC 6D F5 D8 FB DD 4E 5C 7E E5 83 14 4E 72 2D 16 4E 0E 23 6E 4C 52 73 84 25 72 BB 62 D7 CE 23 9E 7F E2 DC A4 A1 C5 80 89 77 A9 61 6C DC A5 D5 C2 99 C0 45 67 DD 4B 39 3F 16 72 FF 76 C0 55 94 F1 B7 E5 73 74 44 B2 57 AE B9 B2 96 99 E4 19 F0 5B A3 CC 30 08 B9 08 5A 7F 97 18 65 25 8D CE 3D 64 2A 81 C0 AA 37 84 B0 F6 00 C2 F2 6E 86 DE B5 B1 73 8C 7F 93 28 4C 0F F8 CD 20 8B E1 45 B7 9A F5 AF B0 4A 6B B3 AD 13 7E 7E A8 A4 CF 02 23 03 99 96 9B 70 75 63 24 D4 BE 23 E7 D7 8A 65 B4 9F D5 68 0A DD F3 20 94 19 69 E4 CB FC 08 CC A4 E5 F9 38 3F 27 4B 8A 69 7A 39 4D F1 08 FE 08 3B F1 73 58 24 49 76 9A EA 20 F6 EB 98 49 03 E4 FE 62 AE 76 70 FB 89 F2 80 B3 81 B7 49 92 46 86 9C B9 4E B9 7A 75 54 08 4C 4D AE DD D1 18 25 5D BF 4D 00 D9 94 70 8A 90 9E 82 C0 C4 D9 DA 21 15 85 AB C3 6A CF 6B DF A5 FC A5 E5 B1 75 AB 8E EA 19 AB D5 CC 68 38 CA FC 1D F9 E7 8E 88 F5 2F E4 57 89 B7 97 78 83 74 B6 4B 9C 79 46 BD D6 90 9E 7E 7E A8 A4 CF 02 23 03 99 96 0D 68 56 0D BD 30 F0 27 97 B1 F2 5A FB 37 E5 00 8D 15 EA 42 0C E0 F5 43 23 20 09 FF 56 88 07 A2 C1 F1 D9 16 87 80 28 66 AF 45 B1 CB BE 68 CF 26 B8 F0 B2 07 89 48 49 0C 81 1A E7 29 39 31 B1 21 59 9F 00 F8 21 15 FD 04 B1 03 87 9B C0 B3 A6 36 7E 46 92 AB 8C F5 E9 27 23 05 14 E5 96 B8 A4 4B A0 A1 A8 19 EF 3B CD B6 03 03 5D 70 DE D8 CE 3E FF B8 AF D3 EA 88 59 78 14 27 AE E4 7F 3A 4C 53 87 F0 59 2F 45 7C 1E 89 E8 A2 6C AE 84 0F 90 0B 0E 05 C1 A2 8D C1 13 6E 99 4E BD BE 7E 7E A0 67 CF 02 23 13 FB F1 F3 EE 35 74 A6 1D 8C 76 17 73 12 F3 A0 75 A8 B9 99 66 58 CA 43 BC 22 B0 72 7A A4 4E 2F 95 2A 58 BC 50 94 2E 6D 2E 91 5D F5 4B B4 6C 56 19 9C 7B 19 80 1C 13 1D D8 C0 2F 21 07 42 4E 88 75 70 5D 95 91 C0 76 C9 11 04 94 C6 9C B6 79 C4 3F 9D 2D B5 AF B5 8 [... cut ...]
Jan 30 21:40:17 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.7.8.2.255 2: 143298
Jan 30 21:40:17 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.0.255 2: 1770207
Jan 30 21:40:17 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.1.255 2: 836079
Jan 30 21:40:17 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.2.255 2: 934128
Jan 30 21:40:17 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.0.13.7.0.255 2: 759

raymar9, 2024-01-31:
Is there any more log after this listing of raw values at 21:40:17? Or does the software just freezes there?
I see the error "ERROR:smartmeter:DLMS data is no list or empty list. Not parsable.". But the software should recover from that. However maybe there is a bug concerning switching to system time as noted in the warning following. I will look into that.

urskle, 2024-01-31:
Yes, there are values after the error. But there are no values in the
influxDB anymore, therefore no data in Grafana.

raymar9, 2024-01-31:
Can you also provide some of the DEBUG log following after 21:40:17?

urskle, 2024-01-31:
Please find annexed the logfile until 21:47.

-- Journal begins at Tue 2024-01-30 20:58:11 CET, ends at Tue 2024-01-30 21:47:19 CET. --

Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:DLMS packet complete and ready for parsing.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:40 0.6.25.9.0.255 1: 0.6.25.9.0.255
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:1 0.0.42.0.0.255 2: ISK1030776617361
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:1 0.0.96.1.1.255 2: 1879668
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:8 0.0.1.0.0.255 2: 01/30/24 21:39:45
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.1.7.0.255 2: 3540
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.2.7.0.255 2: 0
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.3.7.0.255 2: 3103
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.4.7.0.255 2: 0
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.1.8.0.255 2: 27347713
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.1.8.1.255 2: 8137232
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.1.8.2.255 2: 19210481
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.2.8.0.255 2: 15573498
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.2.8.1.255 2: 11723273
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.2.8.2.255 2: 3850225
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.5.8.0.255 2: 10398525
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.5.8.1.255 2: 3413923
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.5.8.2.255 2: 6984602
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.6.8.0.255 2: 657011
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.6.8.1.255 2: 509212
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.6.8.2.255 2: 147799
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.7.8.0.255 2: 565815
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.7.8.1.255 2: 422517
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.7.8.2.255 2: 143298
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.0.255 2: 1770207
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.1.255 2: 836079
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.1.8.8.2.255 2: 934128
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:3 1.0.13.7.0.255 2: 753
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 3540.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 0.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 3103.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 0.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 27347713.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 8137232.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 19210481.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 15573498.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 11723273.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 3850225.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 10398525.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 657011.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 565815.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 1770207.0, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 0.753, "timestamp": 1706647185} sent to MQTT broker.
Jan 30 21:40:02 raspberrypi smartmeter-datacollector[1175]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 8D 2B 92 E7 93 B9 17 94 14 9C E4 C1 8E 27 8F 68 1C DB F5 6F 4B 3D 57 65 62 97 28 A2 4A 46 5B 65 9F 79 71 B1 51 09 11 FB 4A 9D 5D 6E E6 33 F6 CB E9 31 91 44 C8 60 07 B2 CB 74 14 A5 94 7D DC 75 98 B9 A4 D9 1A 1A C7 74 43 7F 78 ED 6D 96 64 C3 35 4D 71 47 4D 1F D7 [... cut ...]
Jan 30 21:40:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 11723273.0, "timestamp": 1706643647} sent to MQTT broker.
Jan 30 21:40:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 3850225.0, "timestamp": 1706643647} sent to MQTT broker.
Jan 30 21:40:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 10398576.0, "timestamp": 1706643647} sent to MQTT broker.
Jan 30 21:40:47 raspberrypi smartmeter-datacollector[1175]: DEBUG:sink:{"value": 657011.0, "timestamp": 1706643647} sent to MQTT broker.
Jan 30 21:

urskle, 2024-01-31:
I just realized that the service influxd fails when no data are displayed
in Grafana. When everything is working, influxd is active and running. I am
trying now to restart influxd with a cronjob when no data are entered in
influxdb.

***@***.***:~ $ sudo systemctl status influxd
● influxdb.service - InfluxDB is an open-source, distributed, time series database
     Loaded: loaded (/lib/systemd/system/influxdb.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-01-31 15:36:49 CET; 1min 37s ago
       Docs: https://docs.influxdata.com/influxdb/
    Process: 900 ExecStart=/usr/lib/influxdb/scripts/influxd-systemd-start.sh (code=exited, status=1/FAILURE)
        CPU: 17ms

Jan 31 15:36:49 raspberrypi systemd[1]: influxdb.service: Scheduled restart job, restart counter is at 5.
Jan 31 15:36:49 raspberrypi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Jan 31 15:36:49 raspberrypi systemd[1]: influxdb.service: Failed with result 'exit-code'.
Jan 31 15:36:49 raspberrypi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
***@***.***:~ $

urskle, 2024-01-31:
Now there is a new error:
Jan 31 16:17:27 raspberrypi telegraf[941]: 2024-01-31T15:17:27Z E!
[outputs.influxdb] When writing to [http://127.0.0.1:8086]: failed doing
req: Post "http://127.0.0.1:8086/write?db=smartmeter": dial tcp
127.0.0.1:8086: connect: connect>
Jan 31 16:17:27 raspberrypi telegraf[941]: 2024-01-31T15:17:27Z E! [agent]
Error writing to outputs.influxdb: could not write any address
This might be helpful to analyze the errors in smartmeter-datacollector

Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.2.8.0.255 2: 15576127
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.2.8.1.255 2: 11725902
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.2.8.2.255 2: 3850225
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.0.255 2: 10416050
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.1.255 2: 3421560
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.2.255 2: 6994490
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.6.8.0.255 2: 657460
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.6.8.1.255 2: 509661
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.6.8.2.255 2: 147799
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.7.8.0.255 2: 566210
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.7.8.1.255 2: 422912
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.7.8.2.255 2: 143298
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.8.8.0.255 2: 1770906
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.8.8.1.255 2: 836588
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.8.8.2.255 2: 934318
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.0.13.7.0.255 2: 0
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 0.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 440.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 0.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 236.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 27371067.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 8146836.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 19224231.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 15576127.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 11725902.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 3850225.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 10416050.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 657460.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 566210.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 1770906.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:17 raspberrypi smartmeter-datacollector[432]: DEBUG:sink:{"value": 0.0, "timestamp": 1706714235} sent to MQTT broker.
Jan 31 16:17:27 raspberrypi telegraf[941]: 2024-01-31T15:17:27Z E! [outputs.influxdb] When writing to [http://127.0.0.1:8086]: failed doing req: Post "http://127.0.0.1:8086/write?db=smartmeter": dial tcp 127.0.0.1:8086: connect: connect>
Jan 31 16:17:27 raspberrypi telegraf[941]: 2024-01-31T15:17:27Z E! [agent] Error writing to outputs.influxdb: could not write any address
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 DB 08 49 53 4B 67 74 91 16 91 82 02 B5 20 00 0C 7E
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 DB 08 49 53 4B 67 74 91 16 91 82 02 B5 20 00 0C 7E 8F B6 EA 36 7B D2 86 B4 A8 0C 6F 92 EA 3F 0F F9 74 DC 14 C3 >
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:HDLC frame incomplete and will not be parsed yet.
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:HDLC Buffer: 7E A8 A4 CF 02 23 03 99 96 E6 E7 00 DB 08 49 53 4B 67 74 91 16 91 82 02 B5 20 00 0C 7E 8F B6 EA 36 7B D2 86 B4 A8 0C 6F 92 EA 3F 0F F9 74 DC 14 C3 >
[... cut ...]
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.2.8.2.255 2: 3850225
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.0.255 2: 10416050
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.1.255 2: 3421560
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.5.8.2.255 2: 6994490
Jan 31 16:17:32 raspberrypi smartmeter-datacollector[432]: DEBUG:smartmeter:3 1.1.6.8.0.255 2: 657460
Jan 31 16:17:32 raspberrypi

urskle, 2024-01-31:
> :
> I think I found a / the problem: when receiving non-parsable data from the
> smartmeter (ERROR in log) the datacollector falls back to system time
> (followup WARNING in log). However if you compare the unix timestamps which
> accompany the data sent to the MQTT broker and therefore the influxDB
> before and after 21:40:17 you detect a backwards time leap about 1 hour. So
> we feed non-monotonic time samples into the database. Somehow the system
> time is not correctly determined in the datacollector service. I'll look
> into that. You would might find the data 1 hour earlier if the influx db
> weren't crashing.
>
> However I'm still not sure why the influxDB crashes. It should handle a
> request from Grafana when no data is available in the requested timespan.
> The error from telegraf (which sits between the MQTT broker and the
> influxDB) probably only appears because of the crashed influxdb which
> telegraf cannot connect to.
>
> First check your time settings (timezone) on the RPi with timedatectl and
> alter them if incorrect.
> A workaround for now that you can try is to manually edit the config file
> stored at /var/lib/smartmeter-datacollector/datacollector.ini by adding systemtime
> = true to the smartmeter config section. This enforces using system time
> from the beginning which mitigates the time leap problem.

Time settings should be ok:
***@***.***:~ $ sudo timedatectl
               Local time: Wed 2024-01-31 17:02:09 CET
           Universal time: Wed 2024-01-31 16:02:09 UTC
                 RTC time: n/a
                Time zone: Europe/Zurich (CET, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

urskle, 2024-02-04:
It still doesn't work. The time settings are correct, I even changed the
time zone to /Europe/Paris. I don't know where to put systemtime = true in
the   /var/lib/smartmeter-datacollector/datacollector.ini file. There are
just sections for reader0, sink0 and logging.
As a workaround I make now a query and a restart for the 2 services influxd
and python3-smartmeter-datacollector when no values are available in the
influxdb. So far, this is functionning.

raymar9, 2024-02-05:
Add `systemtime = true` to the `reader0` section.

urskle, 2024-02-05:
Grafana doesn't display any values with systemtime = true in the
datacollector.ini file in section reader0. Without this entry values are
displayed for a few hours.

raymar9, 2024-08-27:
There has been a new release (v1.2.1) in April concerning an issue with received timestamps from the smartmeter. If you still experience issues, you might try this version.

closed 2024-08-27 by raymar9

urskle, 2024-08-28:
Thank you very much for your hint!
Kind regards, Urs