scs/smartmeter-datacollector #60: LG-E450 wired via RJ12 "MBUS to TTL adapter" to Rpi3B+ UART (GPIO) gives NO DATA

pilorain, 2024-01-30:
LG-E450 was installed in Switzerland by Romande-Energie who gives "Spécifications techniques Landis.pdf" telling data are pushed every 5 seconds on RJ12.

I have bought a TTL to MBUS slave module (1) which is wired from RJ12 to Raspberry Pi 3B+ on GPIO pins 14 & 15 (UART).

A small "serial_read.py" (b) prints a salve of data (c) every 5 seconds according to above specs, so hardware should be OK. Note that Rpi MUST be configure with sudo raspi-config on "3 Interface Options", "I6 Serial Port..." No, Yes to get : "The serial login shell is disabled" & "The serial interface is enabled".

Now I was confident to try the "Demo (Raspberry Pi Image)" but unfortunately it gives NO DATA on Grafana Webinterface.

Thanks in advance for your help.

Steps to reproduce :
1. Download the Raspberry Pi image from the latest release.
2. Flash the image on a (micro)SD card, using the Raspberry Pi Imager.
2.1 Hostname rpi3-lge450
2.2 Enable SSH and use password
2.3 Username pilo and password
2.4 Configure Wi-Fi, SSID, password AND Land for Wi-Fi : CH
2.5 Time zone Europe/Zurich
2.6 Keyboard : ch
3. Once the Raspberry Pi has booted up, I did :
3.1 check SSH access is OK then
3.2 configure serial interface as stated above (raspi-config) and
3.3 launch "serial_read.py" (b) viewing its data (c) and then abort it.
4. Open the configuration web interface at http://192.168.1.116:8000/
5. Adjust the settings of the connected smart meter as explained in Configuration :
5.1 type = lge450
5.2 port =  /dev/serial0
5.3 key =
5.4 MQTT sink is
5.4.1 host = localhost
5.4.2 port = 1883
6. Open Grafana at http://192.168.1.116:3000/
6.1 Home > Dashboards > smartmeter > L+G E450 show "No data" everywhere
7. Issued journalctl --since "10 minutes ago" -u python3-smartmeter-datacollector
See attached files
(a) [journalctl_output.txt](https://github.com/scs/smartmeter-datacollector/files/14103274/journalctl_output.txt)
(b) [serial_read.py.txt](https://github.com/scs/smartmeter-datacollector/files/14103295/serial_read.py.txt)
(c) [serial_read.py.output.txt](https://github.com/scs/smartmeter-datacollector/files/14103343/serial_read.py.output.txt)
Reference
(1) TTL to MBUS, serial port to MBUS slave module, instead of TSS721A, signal isolation
https://www.aliexpress.com/item/32871499543.html?gatewayAdapt=glo2deu

raymar9, 2024-01-31:
The journal log output shows that the serial device cannot be opened. However I cannot see the reason.
Can you provide the config file for the datacollector after you generated it with the web interface? It is stored on the RPi here: `/var/lib/smartmeter-datacollector/datacollector.ini`

pilorain, 2024-01-31:
Here you have [datacollector.ini.txt](https://github.com/scs/smartmeter-datacollector/files/14110051/datacollector.ini.txt) - I did add .txt to attach.

pilorain, 2024-01-31:
Meanwhile I made a change
sudo nano /boot/config.txt
and add (at the end of the file):
dtoverlay=pi3-miniuart-bt
sudo reboot
thanks to https://www.circuits.dk/setup-raspberry-pi-3-gpio-uart/
[journalctl_output.txt](https://github.com/scs/smartmeter-datacollector/files/14116979/journalctl_output.txt)
[datacollector.ini.txt](https://github.com/scs/smartmeter-datacollector/files/14117015/datacollector.ini.txt)
BUT in http://192.168.1.116:8000/ "Restart Demo" gives "Request failed." and
in http://192.168.1.116:3000/d/bcf35a89-ef0d-45e4-9359-cdcbf526ab43/l-g-e450?orgId=1&refresh=1m
there is still NO DATA in Grafana.

raymar9, 2024-02-01:
Since you see the blue bar chart "Zählerstand Tarife" there must be data received by Grafana. Did you check other time spans like "last 2 days" for samples? Maybe it is a timestamp issue.
You can also enable the DEBUG log in the smartmeter-datacollector. This lets you inspect the received data from the meter including the timestamps. You may do this with the web interface or by manually modifying the config file `/var/lib/smartmeter-datacollector/datacollector.ini`, see https://github.com/scs/smartmeter-datacollector/wiki/Configuration#logging-settings.

pilorain, 2024-02-03:
Thanks, I did enable the DEBUG log and I got this kind of data:
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:DLMS packet complete and ready for parsing.
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:40 0.9.25.9.0.255 1: 0.9.25.9.0.255
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:3 1.1.1.8.1.255 2: 1207297
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:3 1.1.1.8.2.255 2: 939710
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:3 1.1.2.8.1.255 2: 556720
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:smartmeter:3 1.1.2.8.2.255 2: 337098
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:sink:{"value": 1207297.0, "timestamp": 1706985480} sent to MQTT broker.
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:sink:{"value": 939710.0, "timestamp": 1706985480} sent to MQTT broker.
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:sink:{"value": 556720.0, "timestamp": 1706985480} sent to MQTT broker.
Feb 03 19:37:59 rpi-lge450 smartmeter-datacollector[1493]: DEBUG:sink:{"value": 337098.0, "timestamp": 1706985480} sent to MQTT broker.

I do not want to struggle with Grafana because next step will be to send smartmeter-datacollector's data to my Home Assistant platform.

The four above values matches the four indexes I can manually read on the LG-E450. So issue as stated in title is fixed. It was not a bug but the way to config serial and UART on Raspberry Pi 3.

pilorain, 2024-10-12:
[SOLVED] It was my hardware mistake ! I have discovered too much later that "**All UARTs on the Raspberry Pi are 3.3V only**". As soon as the wire GPIO_Pin_4__5V was moved to GPIO_Pin_1__3.3V the problem was solved.

Note that meanwhile I have reinstalled Bullseye on my Rpi3B+ ; Home Assistant installed with Docker and then added L&G_E450 smartmeter-datacollector accordind to methode 3.

closed 2024-10-12 by pilorain