scs/smartmeter-datacollector #61: LG-E450 Romande-Energie (Switzerland) data not fully parsed

pilorain, 2024-02-06:
In that case your meter only sends these 4 values about the energy meter readings ("Zählerstand") and no actual power information.
1.1.1.8.1 is imported energy until now in tariff 1 (high-tariff?), 1.1.1.8.2 is same for tariff2 (low-tariff?),
1.1.2.8.1 is exported energy up to now in tariff 1, 1.1.2.8.1 is same for tariff2
If you let HomeAssistant compute the derivative you get a rough estimation of the current power you consume/produce.

Check https://github.com/scs/smartmeter-datacollector/discussions/57 where someone has successfully integrated this project into their HA system.

_Originally posted by @raymar9 in https://github.com/scs/smartmeter-datacollector/issues/60#issuecomment-1926372272_

pilorain, 2024-02-06:
According to [Spécifications techniques Landis .pdf](https://github.com/scs/smartmeter-datacollector/files/14185362/Specifications.techniques.Landis.pdf) the 4 values are those at "Second 0", that is each minute, as viewed in 
[DEBUG_log.ods](https://github.com/scs/smartmeter-datacollector/files/14185560/DEBUG_log.ods). Then at Seconde 5, 10, 15, 20, 25, 30, 35, 40, 45, 50, 55 the log shows eleven times "More DLMS data expected. Not yet ready to be parsed." instead of parsing OBIS frame.
Would it be possible to get those values parsed, at least current power, instead of compute in HA ?

raymar9, 2024-02-07:
I checked your data from your DEBUG logs with the tool [DLMS-Translator](https://www.gurux.fi/gurux_dlms_translator45/Gurux_dlms_translator.html). I usually use this as a reference tool.
The meter uses "GeneralBlockTransfer" to split up the measurement data into chunks. In each chunk the flag `Last block` is `false`, except the last one at the end of each minute. I would expect that a `Last block = true` should be received once every 5s to declare the completion of a measurement series as specified from your energy provider. If you analyze the accumulated data in the buffer right before the DLMS packet could be parsed by the software, it can be observed the following:
1. The data repeats itself (the frames are almost the same),
2. The whole construct is declared invalid by the DLMS translator when concatenating all the PDUs from the GeneralBlockTransfers
3. Parsing only the first block / PDU yields 5 values but with the warning that more values are missing.

I assume therefore that these 5s intervals overburden the smart meter and it is always only able to send the first block of a measurement series. The top of minute measurement series is shorter and due to that the meter can send it complete and valid.
Actually, EKZ mentioned this problem about the L+G E450 meter. They decided to prolong the interval to 15s which seems to work.

In summary: No, we unfortunately can not do anything about that on the receiver side because the received data is invalid. Either you live with that (use the method with HA) or present these findings to your energy provider and ask them to change the configuration.

closed 2024-02-07 by raymar9

pilorain, 2024-10-12:
See https://github.com/scs/smartmeter-datacollector/issues/60#issuecomment-2408642888