scs/smartmeter-datacollector #62: Raspberry Pi image doesn't boot

gitso7, 2024-03-06:
I try to run the provided image on Raspberry Pi Zero 2 W (
Downloading the image several times, writing it to different sd card always leads to the same problem: the Raspberry doesn't boot up. It hangs at the (multicolored) splash screen and the LED on the Raspberry flashes repeatedly 7 time (aka "kernel. img not found").
Any suggestions?

mateblue, 2024-06-11:
Hi gitso7
I used the smartmeter-datacollector-demo_v1.2.1_rpi-image-lite.zip and the imager_1.8.5 with my rpi3 and it worked just fine.
Which Versions did you use? Do other images work?

cavalon, 2024-06-24:
Using a Raspberry Pi Zero 2 W Rev 1.0 for 2+ years without any issue...

What did you write your SD card with?

raymar9, 2024-08-05:
We tested the image (v1.2.1) on a Raspberry Pi Zero 2 W last week. It booted successfully. The SD-card was flashed with the official "Raspberry Pi Imager" tool.

closed 2024-08-05 by raymar9