scs/smartmeter-datacollector #67: M&C SX401 G3 Support

aschlapbach, 2024-04-21:
After few years with an E450, Zurich EKZ has just replaced it with a new smart meter type:
https://www.meterandcontrol.com/products/ami-devices/sx401-g3-plc/

It was installed with 4 lead seals.

Any clue how this project supports this type?

And why EKZ updates hardware after few years?

raymar9, 2024-05-02:
I can only answer the first question: Unfortunately, this project does not support the "Meter & Control" smart meter and it is not planned at the moment to add this device.
For other questions you need to contact the EKZ customer support.

closed 2024-05-02 by raymar9

aschlapbach, 2024-05-03:
EKZ just wrote:
"Leider ist bei ihrem Smart-Meter keine Aktivierung der lokalen Schnittstelle möglich.
Eine detaillierte Begründung finden sie in unserem «Fragen-Antworten-Katalog» unter der Frage: Für welche Kunden ist die lokale Schnittstelle nicht verfügbar und warum?
Weitere Informationen finden sie in unserem Fragen-Antworten-Katalog:
https://www.ekz.ch/dam/ekz/myekz/FAQ-Nutzung-lokale-Schnittstelle-am-SM.pdf
"