scs/smartmeter-datacollector #68: Forward also other OBIS codes for i.e.  water

mateblue, 2024-06-11:
Hi 
My Electricity Provider sends beside the power meter information also the water meter reading.
On the MBUS I get the obis codes
0.1.96.1.0 : Water meter ID
0.1.24.2.1 : Water meter reading
It would be nice to have this counter also sent to the sinks.

Here the logs:
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 04 15 7A 0C 07 E8 06 0B 02 10 00 00 FF 80 00 00 02 0A 01 0A 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 0B 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 01 60 01 00 FF 0F 02 12 00 00 02 04 12 00 01 09 06 00 02 60 01 00 FF 0F 02 12 00 00 02 04 12 00 01 09 06 00 03 60 01 00 FF 0F 02 12 00 00 EB F5 7E 7E A0 86 CE FF 03 13 9A 9D E0 40 00 02 00 00 75 02 04 12 00 01 09 06 00 04 60 01 00 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 01 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 02 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 03 18 02 01 FF 0F 02 12 00 00 02 04 12 00 04 09 06 00 04 18 02 01 FF 0F 02 12 00 00 09 06 00 0B 19 09 00 FF 09 08 32 34 35 32 31 36 36 32 09 01 00 09 01 00 09 01 00 C4 FD 7E 7E A0 25 CE FF 03 13 92 6A E0 C0 00 03 00 00 14 05 00 00 77 51 05 00 00 00 00 05 00 00 00 00 05 00 00 00 00 87 59 7E
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:DLMS packet complete and ready for parsing.
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:40 0.11.25.9.0.255 1: 0.11.25.9.0.255
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:**1 0.1.96.1.0.255 2: 24521662**
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:1 0.2.96.1.0.255 2:
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:1 0.3.96.1.0.255 2:
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:1 0.4.96.1.0.255 2:
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:4 **0.1.24.2.1.255 2: 30545**
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:4 0.2.24.2.1.255 2: 0
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:4 0.3.24.2.1.255 2: 0
Jun 11 16:00:01 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:4 0.4.24.2.1.255 2: 0
Later it does also send:
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:HDLC Buffer: 7E A0 84 CE FF 03 13 12 8B E6 E7 00 E0 40 00 01 00 00 70 0F 00 04 15 7D 0C 07 E8 06 0B 02 10 00 05 FF 80 00 00 02 0C 01 0C 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 02 12 00 00 02 04 12 00 28 09 06 00 08 19 09 00 FF 0F 01 12 00 00 02 04 12 00 01 09 06 00 00 60 01 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 01 07 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 00 02 07 00 FF 0F 02 12 00 00 2C 55 7E 7E A0 7D CE FF 03 13 D0 45 E0 40 00 02 00 00 6C 02 04 12 00 03 09 06 01 01 01 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 02 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 05 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 06 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 07 08 00 FF 0F 02 12 00 00 02 04 12 00 03 09 06 01 01 08 08 00 FF 0F 02 12 00 00 C1 95 7E 7E A0 62 CE FF 03 13 6C 9B E0 C0 00 03 00 00 51 02 04 12 00 01 09 06 00 00 60 03 00 FF 0F 02 12 00 00 09 06 00 08 19 09 00 FF 09 08 36 31 36 39 33 30 37 35 06 00 00 06 7D 06 00 00 00 00 06 00 0D 3D 8C 06 00 12 25 09 06 00 00 00 47 06 00 00 00 00 06 00 06 7B 47 06 00 07 C9 FF 06 00 00 00 0C EE 1B 7E
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:DLMS packet complete and ready for parsing.
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:40 0.8.25.9.0.255 1: 0.8.25.9.0.255
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:1 0.0.96.1.0.255 2: 61693075
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.0.1.7.0.255 2: 1661
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.0.2.7.0.255 2: 0
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.1.8.0.255 2: 867724
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.2.8.0.255 2: 1189129
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.5.8.0.255 2: 71
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.6.8.0.255 2: 0
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.7.8.0.255 2: 424775
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:3 1.1.8.8.0.255 2: 510463
Jun 11 16:00:07 smartmeter smartmeter-datacollector[217978]: DEBUG:smartmeter:1 0.0.96.3.0.255 2: 12

raymar9, 2024-06-21:
Hi mateblue
These OBIS-codes for your water meter are rather specific for your setup and provider, so unfortunately I cannot integrate them into the project for everyone. For example: the first number `0` means "abstract medium" in the OBIS standard, however water would actually be `8`. `24.2` just means "M-Bus value" which could be anything.

However, you can clone (or even fork) the project and extend the software yourself. I try to give you some hints where to start:

- In https://github.com/scs/smartmeter-datacollector/blob/master/smartmeter_datacollector/smartmeter/meter_data.py add new  `MeterDataPointType` for the water meter reading to the enum `MeterDataPointTypes`. For example:
`WATER_VOLUME = MeterDataPointType("WATER_VOLUME", "Water volume used", "m3")`
- In https://github.com/scs/smartmeter-datacollector/blob/master/smartmeter_datacollector/smartmeter/cosem.py add new `RegisterCosem` for your received OBIS code and previous created `MeterDataPointType`. For example:
`RegisterCosem(OBISCode(0, 1, 24, 2, 1), MeterDataPointTypes.WATER_VOLUME .value)`

mateblue, 2024-06-23:
Hi Mario
Thanks for your reply and thanks for your hints :-)
I did fork the project now and made my changes as good as I could....

Do you know why there are two ways used to get the register value? The obj.getValues()[1] works fine with me but _extract_register_value(register: GXDLMSRegister) -> Optional[Any]:
        return register.getValues()[1]
did not work for my EXTENDED REGISTER (what my water counter is on).

Is that the reason why all other objectTypes then REGISTER are filtered out? 

Thanks for all the work in smartmeter-datacollector. It's verry nice to have it open-source and to be able to custonice it as needed.

Cheers

Guido
Guido

raymar9, 2024-06-26:
Hi Guido
You're welcome :)
Can you specify more clearly where `obj.getValues()[1]` is used (link to code line & file)? I found https://github.com/scs/smartmeter-datacollector/blob/master/smartmeter_datacollector/smartmeter/cosem.py#L134 which is only used to retrieve the ID of the meter. 

And what do you mean by "EXTENDED REGISTER"?

There are multiple ways data is sent by smart meters (which are configured by the providers in different ways). Some use PushLists others don't. We try to support both options if the required information is sent (OBIS codes & corresponding data).

mateblue, 2024-06-26:
Hi Mario
This log line shows "obj.objectType" and "obj.getValues()[attr_ind - 1]" 
https://github.com/scs/smartmeter-datacollector/blob/209848b50a2af461688dea69b94f077cd90a2e89/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py#L164
For me it was: "DEBUG:smartmeter:**4** 0.1.24.2.1.255 2: **30545**", so reading out the raw value did work. 

This line does work for normal "REGISTER" but for me the "EXTENDED REGISTER"s result in raw_value = 0
https://github.com/scs/smartmeter-datacollector/blob/209848b50a2af461688dea69b94f077cd90a2e89/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py#L146
I replaced it with "raw_value = obj.getValues()[1]" since this worked for the log line above. Ok, [1] should be the right attr_ind... but in my case it is always 1 :-)

The first value in the log line is 4 - obj.objectType - which I think is gurux EXTENDED_REGISTER
https://github.com/Gurux/Gurux.DLMS.Python/blob/848400a75c5cd60469a67ce13d812589e82d7680/Gurux.DLMS.python/gurux_dlms/enums/ObjectType.py#L110

Hope that helps to understand what I ment.

The filter here mean that only Gurux ObjectType.REGISTER (so only obj.objectType 3) are taken (and not EXTENDED_REGISTER (4) nor others) so I had to take the filter out.
https://github.com/scs/smartmeter-datacollector/blob/209848b50a2af461688dea69b94f077cd90a2e89/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py#L143

I took also the "and isinstance(obj, GXDLMSRegister)" out since I think that my EXTENDED_REGISTER obj is an GXDLMSExtendedRegister object.
https://github.com/scs/smartmeter-datacollector/blob/209848b50a2af461688dea69b94f077cd90a2e89/smartmeter_datacollector/smartmeter/hdlc_dlms_parser.py#L145

My Fork which is productive running for me is here: 
https://github.com/mateblue/smartmeter-datacollector/

Thanks a lot for your work

Guido

raymar9, 2024-07-08:
Hey Guido
I looked into this topic a bit further. You are right, the EXTENDED_REGISTER which is a DLMS-protocol feature are currently not supported by our software. Up to now, we had no example meter with extended registers. At the same time I noticed an issue/bug in our code when parsing multiple DLMS objects with the same OBIS-code but different channel number (2nd position, B, in OBIS codes).

I created a [new branch](https://github.com/scs/smartmeter-datacollector/tree/feature/68_parse-extended-registers) with the EXTENDED_REGISTER extension and the [bugfix](https://github.com/scs/smartmeter-datacollector/commit/81cf4e677b29deecdd47cbca7cd7b3e12beace51). Our software now still does only accept one channel per OBIS code, but with the fix the register of the first occurrence is taken and won't be overwritten. That was probably another issue why your value was 0 or empty.

I added a unit test with your data sample above.
Thanks for your explanations!

mateblue, 2024-07-08:
Hi Mario
Thanks for your new branch. 
I tested it and it passt my usecase here. My water volume values do come thru now.

I have two other OBIS Codes which I send to the sink at the moment.
One is 1 0.0.96.3.0.255 - Which is the tariff 
and 1 0.1.96.1.0.255 - Which is the water counter ID.
Both are objet type 1 = DATA therefore they get filtered out in your code.
So I stay on my fork for the moment.

1 0.0.96.3.0.255 - Which is the tariff :
My smartmeter does not (yet?) send me the 
- ACTIVE_ENERGY_P_T1
- ACTIVE_ENERGY_P_T2
- ACTIVE_ENERGY_N_T1
- ACTIVE_ENERGY_N_T2 
so I do not have any data regarding the tariff - the only thing I have is 1 0.0.96.3.0.255  which shows me the tariff at point in time.
With this value I can calculate my energy per tariff and I can automate stuff regarding tariff.

1 0.1.96.1.0.255 - Which is the water counter ID:
I am not sure which meterID the water volume should get. At the moment it does get the meterID of the smartmeter which sends the data. 
Since I don't want to loose the water counter ID I send it also to the sink. 
What do you think? Should the EXTENDED_REGISTERs get its own meterIDs?

Thanks

Guido

raymar9, 2024-08-27:
Sorry for the late response, Guido.
Currently the software is based on one meter ID per attached device / CII. The usecase with a secondary meter attached to the smart meter with a separate ID whose measurements get somehow "extracted" from the main datapoints would require some fundamental changes to the software. If that would follow a well-defined standard we could implement this. However, it might be possible that with a different meter and provider transmitting a second meter ID and register data could be configured differently.
Therefore, we stay with the current compromise of parsing EXTENDED_REGISTERS. I hope you understand.

closed 2024-08-27 by raymar9