scs/smartmeter-datacollector #74: Failed to extract data from HDLC frame: 'Invalid data type.' Some data got lost

M1dfsdiekfj, 2024-11-03:
Hi there

I want to get access to my smart meter (L&G E450, installed in TG/CH). I use method 1 and 3 on different RPIs, both with the same results. After I changed successfully the baud rate in the py-file to 9600, the LED/USB-Reader actually receives some data. However, based on the syslog file (e.g. from the RPI with demo/method 1) I encounter every couple of minutes this error:

> smartmeter smartmeter-datacollector[413]: WARNING:smartmeter:Failed to extract data from HDLC frame: 'Invalid data type.' Some data got lost.
 
This shows up directly after the data it receives:
> Nov  3 19:03:44 smartmeter smartmeter-datacollector[413]: DEBUG:smartmeter:HDLC Buffer: 7E A0 6D CE FF 03 13 90 F1 E6 E7 00 0F 00 08 FD 7B 0C 07 E8 0B 03 07 13 00 0F FF 80 00 00 02 02 06 01 41 0C 28 06 00 00 00 00 E6 E7 00 0F 00 08 FD 7C 0C 07 E8 0B 03 07 13 01 0F FF 80 00 00 02 02 06 01 41 0C 3A 06 00 00 00 00 E6 E7 00 0F 00 08 FD 7D 0C 07 E8 0B 03 07 13 02 0F FF 80 00 00 02 02 06 01 41 0C 4B 06 00 00 00 00 FD F2 7E

If it helps, find attached the full syslog file. Hopefully, you can help because for a long time I'v already tried unsuccessfully to get it running to analyse my smart meter data…

Best regards,
Markus 

[Smartmeter LogFile.log](https://github.com/user-attachments/files/17611646/Smartmeter.LogFile.log)

mateblue, 2024-11-04:
Hi Markus
I successfully run a smartmeter-datacollector with my L&G E450 and a USB to MBUS readconverter on a RPI in Rüti ZH.
Do you use a Decryption Key?
Did you configure the smartmeter-datacollector useing the web based configurator page?
Why did you change the baudrate to 9600? Didn't you get anything with 2400?
What is your VNB (EW)? and did you get any documentation? Does it say the baudrate? How freqent should you get which data? 
In your log there are every 3 Minutes a single Telegram.
You can translate your HDLC Messages useing the Gurux Translater:  https://www.gurux.fi/GuruxDLMSTranslator
Then you get something like that: (it does not contain any OBIS codes - so it is probably not the kind of telegrams your looking for...)
<HDLC len="108" >
<TargetAddress Value="13311" />
<SourceAddress Value="1" />

<FrameType Value="13" />
<PDU>
<x:data-notification>
  <x:long-invoke-id-and-priority>589167</x:long-invoke-id-and-priority>
  
  <x:date-time>07E80B030712300FFF800000</x:date-time>
  <x:notification-body>
    <x:data-value>
      <x:structure>
        <x:double-long-unsigned>21039963</x:double-long-unsigned>
        <x:double-long-unsigned>0</x:double-long-unsigned>
      </x:structure>
    </x:data-value>
  </x:notification-body>
</x:data-notification>
</PDU>
</HDLC>
 
Regards

Guido

M1dfsdiekfj, 2024-11-05:
Hi Guido

> Do you use a Decryption Key?

No, it‘s unencrypted. 

> Did you configure the smartmeter-datacollector useing the web based configurator page?

I’ve tried both, manually and webpage based.

> Why did you change the baudrate to 9600? Didn't you get anything with 2400?

That‘s the only value where I get at least some meaningful data which opens and closes with 7E. On 2400 I don‘t get any data. Although having asked them multiple times, I didn’t get any documentation by my VNB (Kierzek AG, Kreuzlingen region, TG).

In general, the data that is sent seems to be meaningful as it _can_ be translated. But the result looks confusing both to me and the Smartmeter-Datacollector…

Best regards,
Markus

raymar9, 2024-11-05:
Hi Markus
The baudrate of 9600 is correct in your case since you receive valid HDLC frames with this setting.
However, unfortunately your smart meter is configured in a way the toolkit is not able to parse the receiving data. Like Guido I tested your data with the handy DLMS translator tool from Gurux.
```
<FrameType Value="13" />
<PDU>
<DataNotification>
  <!-- Invoke ID: 588732 -->
  <LongInvokeIdAndPriority Value="588732" />
  <!-- 03.11.2024 11:33:15 -->
  <DateTime Value="07E80B03070B210FFF800000" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="2" >
        <UInt32 Value="21026017" />
        <UInt32 Value="0" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
</PDU>
```

1. The DLMS data is not long - there can't be a lot of information inside. 
2. The only value the translator tool could retrieve is "21026017". Do you have any idea, when comparing to the shown values on the display of the meter, what this value could be? Used energy? I tested multiple consecutive messages of yours, they rise with each datagram: 21025913, 21025941, 21025974, 21026017, ...
3. The messages do not seem to have any indication what type the values correspond to. Typically, this comes in form of an OBIS code. I once saw another smart meter that acted similarly. Unfortunately, the datacollector is not able to parse this kind of message because it needs to map the received values to a type/OBIS-code. 

Your provider would need to reconfigure your meter. If you're still interested in retrieving this single, unknown value for some purpose, you might write a small script to extract it.

M1dfsdiekfj, 2024-11-05:
Thank you two for delving into that. I looked deeper into what is shown on display by my smart meter yet I didn‘ find out what the value around 210XXXX could mean. So I once again contacted my provider although they themselves seem to have no idea why it doesn't work properly – even after having contacted someone from Landis & Gyr…

closed 2024-11-06 by raymar9