scs/smartmeter-datacollector #76: MQTT service does not start after reboot in demo

raymar9, 2025-01-08:
Hi,
I’m experiencing an issue with Meter. When I reboot the Raspberry Pi (RPI), the MQTT service doesn’t work. To fix this, I have to log in to the web interface and manually press "Restart Data Collector." After that, MQTT starts working again. However, the problem reoccurs every time I reboot the RPI.   It is the demo image.

Best regards,
Miha

_Originally posted by @TTNIdrija in https://github.com/scs/smartmeter-datacollector/issues/70#issuecomment-2575686622_

raymar9, 2025-01-08:
Do you mean with "MQTT service" the `smartmeter-datacollector` service or `mosquitto`, the MQTT broker?

Could you check whether the systemd services for `python3-smartmeter-datacollector` and `mosquitto` are enabled when you access the RPi via SSH?

```
systemctl status python3-smartmeter-datacollector
systemctl status mosquitto
```

You can enable a service with `systemctl enable python3-smartmeter-datacollector` or `systemctl enable mosquitto`. 
Enabling means that it starts during boot up.

TTNIdrija, 2025-01-08:
I solved it by creating a restart service that runs 10 seconds after the Raspberry Pi boots.

`sudo nano /etc/systemd/system/reset-smartmeter.service`

content:
*******************************************************************************************************
```
[Unit]
Description=Reset Smartmeter Data Collector Service
After=network.target

[Service]
Type=oneshot
ExecStart=/bin/bash -c "sleep 10 && /usr/bin/systemctl restart python3-smartmeter-datacollector"

[Install]
WantedBy=default.target
```
******************************************************************************************************
`sudo systemctl daemon-reload`

`sudo systemctl enable reset-smartmeter.service`

`sudo reboot`

Probably not the ideal solution, but it works :)

raymar9, 2025-01-09:
Nice! But why didn't `systemctl enable python3-smartmeter-datacollector` suffice? Did the service start but didn't produce any output or did the service just not start after booting?

I will check the startup behavior again the next time I use the demo image. However, I'm pretty sure that after configuration the demo automatically started and worked after rebooting, the last time I used the demo.