scs/smartmeter-datacollector #78: Iskra AS550

TTNIdrija, 2025-02-10:
Hello,
Does anyone have instructions on how to connect the Iskra AS550 meter to EBUS? My electricity company replaced my Landis electric meter with a new Iskra AS550. I have a USB-to-EBUS adapter for the Landis meter.

Regards,
Miha

Iskra data output

[image]

raymar9, 2025-02-13:
Hi Miha
Is it possible that you have a Iskra AM550 meter and not AS550? And you probably mean a USB-to-MBus adapter?

For the Iskra AM550 you require a P1-to-USB adapter - MBus is not supported by this meter. 
Then, check whether the meter sends its data via DLMS/Cosem (in binary format) and not in DSMR (ASCII) which is not yet supported by the data-collector.

You can find more information here: https://github.com/scs/smartmeter-datacollector/wiki

TTNIdrija, 2025-02-13:
Thanks!
I order new adapter.

closed 2025-09-02 by raymar9