scs/smartmeter-datacollector #81: Debian packages not working with current Debian release

wirr00, 2025-08-30:
Currently Debian stable is v13 "Trixie". Unfortunately the provided "python3-smartmeter-datacollector" package can not be installed with it:
```
Setting up python3-smartmeter-datacollector (1.2.1-1) ...
Adding group smartmeter....done
Adding system user smartmeter....done
Installing dependencies using pip..dpkg: error processing package python3-smartmeter-datacollector (--install):
 installed python3-smartmeter-datacollector package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3-smartmeter-datacollector
```

What is needed to continue using the smartmeter-datacollector with current Debian releases?

kreutpet, 2025-12-30:
hi , 
i am facing the same issue as i move to latest version of debian. 
there is still an issue to install the python pip with the current permissions

`
apt install ./smartmeter-datacollector_1.2.1-1_amd64.deb 
Note, selecting 'smartmeter-datacollector' instead of './smartmeter-datacollector_1.2.1-1_amd64.deb'
Installing:
  smartmeter-datacollector

Summary:
  Upgrading: 0, Installing: 1, Removing: 0, Not Upgrading: 0
  Download size: 0 B / 771 kB
  Space needed: 855 kB / 4’435 MB available

Get:1 /root/smartmeter-datacollector_1.2.1-1_amd64.deb smartmeter-datacollector all 1.2.1-1 [771 kB]
Selecting previously unselected package smartmeter-datacollector.
(Reading database ... 23892 files and directories currently installed.)
Preparing to unpack .../smartmeter-datacollector_1.2.1-1_amd64.deb ...
Unpacking smartmeter-datacollector (1.2.1-1) ...
Setting up smartmeter-datacollector (1.2.1-1) ...
Adding group smartmeter....done
Adding system user smartmeter....done
Adding user smartmeter to group dialout....done
Creating directory /var/lib/smartmeter-datacollector....done
We renamed asyncio-mqtt to aiomqtt and released a version 1.0.0 in the process. This is the last release under the asyncio-mqtt name. You can find the new repository at https://github.com/sbtinstruments/aiomqtt
Traceback (most recent call last):
  File "/usr/bin/smartmeter-datacollector/_bootstrap/__init__.py", line 76, in import_string
    module = __import__(module_name, None, None, [obj_name])
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/smartmeter_datacollector/__main__.py", line 8, in <module>
    from smartmeter_datacollector.app import main
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/smartmeter_datacollector/app.py", line 14, in <module>
    from . import config, factory
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/smartmeter_datacollector/factory.py", line 17, in <module>
    from .smartmeter.iskraam550 import IskraAM550
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/smartmeter_datacollector/smartmeter/iskraam550.py", line 13, in <module>
    from .cosem import Cosem
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/smartmeter_datacollector/smartmeter/cosem.py", line 14, in <module>
    from gurux_dlms import GXDateTime
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/__init__.py", line 49, in <module>
    from .GXDLMSClient import GXDLMSClient
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/GXDLMSClient.py", line 35, in <module>
    from .GXDLMSSettings import GXDLMSSettings
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/GXDLMSSettings.py", line 38, in <module>
    from .objects.GXDLMSObjectCollection import GXDLMSObjectCollection
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/objects/__init__.py", line 52, in <module>
    from .GXDLMSAssociationLogicalName import GXDLMSAssociationLogicalName
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/objects/GXDLMSAssociationLogicalName.py", line 41, in <module>
    from .GXDLMSObjectCollection import GXDLMSObjectCollection
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/objects/GXDLMSObjectCollection.py", line 38, in <module>
    from .GXXmlWriter import GXXmlWriter
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/objects/GXXmlWriter.py", line 38, in <module>
    from ..GXDLMSConverter import GXDLMSConverter
  File "/var/lib/smartmeter-datacollector/.datacollector_cache/smartmeter-datacollector.pyz_b8d92e9228b0de8f3f2b9cf7f4d128874b0f5f0ccbdb9e2f86f0b3cedc6d16ad/site-packages/gurux_dlms/GXDLMSConverter.py", line 35, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/smartmeter-datacollector/_bootstrap/__init__.py", line 84, in import_string
    return getattr(module, obj_name)
AttributeError: module 'smartmeter_datacollector' has no attribute '__main__'. Did you mean: '__name__'?

[... cut ...]
Failed to extract the .pyz file
dpkg: error processing package smartmeter-datacollector (--configure):
 installed smartmeter-datacollector package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 smartmeter-datacollector
Notice: Download is performed unsandboxed as root as file '/root/smartmeter-datacollector_1.2.1-1_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
Error: Sub-process /usr/bin/dpkg returned an error code (1)
`

kreutpet, 2026-01-11:
thanks ... this version work
many thanks for the great work you are doing..

raymar9, 2026-01-16:
We finally created a new release v1.3.0 with updated software. The release is based on Debian bookworm, however the deb packages and Python software should also work with Debian trixie.